openhwfoundation/programs #4: Which Instruction Fetch does CV32E40P and CV32E40 use?

MikeOpenHWGroup, 2019-12-23:
Section 2 of the RI5CY [user manual](https://github.com/pulp-platform/riscv/blob/master/doc/user_manual.doc) states: 
```
There are two prefetch flavors available:
* 32-Bit word prefetcher. It stores the fetched words in a FIFO with three entries.
* 128-Bit cache line prefetcher. It stores one 128-bit wide cache line plus 32-bit to allow for cross-cache line misaligned instructions.
```
Several questions arise:
1. Are these mutually exclusive logic modules?  If so:
* How are they selected at simulation compile-time and synthesis time?
* Which of these are being used by CV32E40P and CV32E40?
2. If these are not mutually exclusive, how does the prefetch unit logic determine which will be used to prefetch from the cache and why?

davideschiavone, 2020-01-11:
According to the meeting on Jan 07
673830ac097d79c6197262eb0cf5e9a783e602a5

The prefetcher will only be the 32bit one.
Keep the issue open to update documentation

Silabs-ArjanB, 2020-06-16:
Hi @MikeOpenHWGroup The User Manaul has been updated (and obsolete references to the 128-bit prefetcher have been removed). Can you please close this issue?

MikeOpenHWGroup, 2020-06-16:
Looks good.  Thanks!

closed 2020-06-16 by MikeOpenHWGroup