openhwfoundation/programs #11: Comments on [...]core-v-docs/tree/master/verif/README.md

alfredoh1234, 2020-01-30:
### Type
* Other: feedback on current file and suggestions for edits

### Location of Issue
https://github.com/openhwgroup/core-v-docs/tree/master/verif/README.md

### Additional context
**[Requirement Location]**
* It would be preferable to have table with names, URL and format used to identify requirement location in tables, to help when combining and sorting spreadsheets when needed

**[Feature]** 
* It may be preferable to use the section header's name in the reference document. This will help when navigating across documents and test plans

**[Sub Feature}**
* This should be mandatory, using the instruction pseudo code as the sub feature, e.g. in the example of the ADDI instruction above:
   addi  rD, rs1, Imm

**[Feature Description]**
* The description should use the Sub Feature pseudo code terms when describing the feature and explain the terms using that perspective. It may help by adopting the defacto description of an instruction.
The four initial columns of the Test Plan may show the ADDi instruction in this way:

Requirement   Location | Feature | Sub-Feature | Feature Description
-- | -- | -- | --
RISC-V Unprivileged ISA Ver. 20190608-Base-Ratified 2.4 | Integer Computational Instructions | addi  rD, rs1, Imm| rD receives the addition of rs1 and Imm

**

MikeOpenHWGroup, 2020-01-30:
Good feedback Alfredo, thanks.   In the future, Please explicitly assign any Issues you generate.   The person who is assigned is responsible for moving the Issue forward.

To address your specific points:

1. **Requirement Location**:  Things are a little too fluid for that now to be included in the Vplans.  The [introduction](https://github.com/openhwgroup/core-v-docs/blob/master/cores/cv32e40p/introduction.rst) document of core has this list of URLs and for now it makes sense to keep it there.
2. **Feature**: Good idea. I updated the [HOWTO](https://github.com/openhwgroup/core-v-docs/edit/master/verif/README.md).
3. **Sub Feature**:   Its not practical to make this mandatory.  There willalso be _one_ feature that does not need it.  That's whythe wording in the [HOWTO](https://github.com/openhwgroup/core-v-docs/edit/master/verif/README.md) is "This is an optional, but often used column.".
4. **Feature Description**: what you are suggesting makes sense, but should not be enforced as a rule.  Every Feature is different and not all will be instructions described with pseudo-code.

alfredoh1234, 2020-01-31:
I would like the Requirement Location to be reconsidered. For example, the table in the PDF document is out of date. version 4.2 should be used, and the link in the introduction mark-down document is not tied to a specific version of the RI5CY user manual. 
More rigour is required for the requirement location

MikeOpenHWGroup, 2020-02-03:
Is the PDF you are referring to the [CV32E40 Features & Parameters](https://github.com/openhwgroup/core-v-docs/blob/master/cores/CV32E40_Features_Parameters.pdf) document?  We should not use this as a reference document since it was not created by Davide, and Davide does not maintain it.  (This was no immediately clear to me either.)  It was/is intended as a discussion document only.

I recommend that we use the documents listed in the [CV32E40P Introduction](https://github.com/openhwgroup/core-v-docs/blob/master/verif/README.md#requirement-location).  Keep in mind that the RI5CY user manual is a work in progress and we are not yet ready to designate a specific version since it will be changing often.  So if the user manual is used, we will need to reference the release/version info in the Vplan.

I have updated the relevant section of the [HOWTO](https://github.com/openhwgroup/core-v-docs/tree/master/verif#requirement-location) to indicate this, have a look.

alfredoh1234, 2020-03-03:
Reviewed changes and comments. Looks good!

closed 2020-03-03 by alfredoh1234