openhwfoundation/programs #14: Vplan for XPULP SIMD Extended Instructions

MikeOpenHWGroup, 2020-02-05:
XPULP SIMD instructions have been categorized as an "Unconditional Required" feature for CV32E40P.

These instructions are documented in Section 14.5 of the  [RI5CY User Manual](https://github.com/pulp-platform/riscv/blob/master/doc/user_manual.doc).  A couple of things to be aware of:

- Its possible that the RISCY User Manual may be transferred to the OpenHW Group and be re-branded as the CV32E40P User Manual during the time this Vplan is captured.

- There is a high probability that the User Manual does not contain sufficient information to complete the SIMD Vplan.  In these instances, create a GitHub issue with a `documentation` label and assign it to Davide Schiavone.

Please use the [Vplan template](https://github.com/openhwgroup/core-v-docs/blob/master/verif/CV32E40P/VerificationPlan/CORE-V_VerifPlan_Template.xlsx) and review the [HOWTO](https://github.com/openhwgroup/core-v-docs/tree/master/verif#how-to-write-a-verification-plan-testplan) document prior to starting this task.

The Vplan spreadsheet shall be committed to the [XPULP Instructon Extensions](https://github.com/openhwgroup/core-v-docs/tree/master/verif/CV32E40P/VerificationPlan/xpulp_instruction_extensions) directory.

alfredoh1234, 2020-03-03:
XPULP packed-SIMD test plan submitted and merge request done.

MikeOpenHWGroup, 2020-03-03:
Nice.  Thanks.  Will close the issue.

closed 2020-03-03 by MikeOpenHWGroup