openhwfoundation/programs #18: Vplan for Debug & Trace

MikeOpenHWGroup, 2020-02-05:
Capture the Verification Plan for both the `Debug` and `Trigger module` features of the CV32E40P.

At this time the [RI5CY User Manual](https://github.com/pulp-platform/riscv/blob/master/doc/user_manual.doc) does not contain sufficient information to capture a Vplan for all of these Features, but it is believed that the [RISC-V External Debug Support Version 0.13.2](https://riscv.org/specifications/debug-specification) specification from the RISC-V foundation does.  If this is not the case, and additional information is required, create a documentation issue and assign it to Davide Schiavone.

### Background information
Please review the [VPLAN HOWTO](https://github.com/openhwgroup/core-v-docs/tree/master/verif#how-to-write-a-verification-plan-testplan) document prior to starting this task.

### Document Template
[Verification Plan template](https://github.com/openhwgroup/core-v-docs/blob/master/verif/CV32E40P/VerificationPlan/CORE-V_VerifPlan_Template.xlsx)

### Document Location
Please commit the Vplan spreadsheet to the [Verification Plan Debug-Trace](https://github.com/openhwgroup/core-v-docs/tree/master/verif/CV32E40P/VerificationPlan/debug-trace) directory.

alfredoh1234, 2020-03-05:
I'll start working on this on the week of March 9

MikeOpenHWGroup, 2020-10-08:
This Vplan has been updated and reviewed.  Closing the issue.

closed 2020-10-08 by MikeOpenHWGroup