openhwfoundation/programs #28: Update IRQ mode description

zetalog, 2020-04-10:
## Task Title
Update Interrupt description and related CSR

### Background information
If the changeset is merged: https://github.com/openhwgroup/cv32e40p/pull/299,
RI5CY User Manual (https://github.com/pulp-platform/riscv/blob/master/doc/user_manual.doc) need to be modified. Especially, section 10.7, 10.8, 10.9 and 12.1.

### Additional context
Reference: https://github.com/openhwgroup/cv32e40p/pull/299

Silabs-ArjanB, 2020-06-16:
Hi @zetalog I have updated the documentation (see https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/exceptions_interrupts.html) with description of the IRQ mode. Could you please close this issue if you think the updated documentation is adequate?

closed 2020-06-17 by davideschiavone

zetalog, 2020-07-26:
Sorry for the delayed response.
I've been busy working on other things these days.

Best regards