openhwfoundation/programs #31: How to handle new interrupt inputs

MikeOpenHWGroup, 2020-04-25:
The CV32E40P has a number of new interrupt inputs:
[image]
Is it safe to tie all these off to 'b0 until such time as the testbenches can do something interesting with them?

Related question: when/where will the function of these new inputs be documented?

Silabs-ArjanB, 2020-04-26:
The testbench could check the following:

- relative priorities for all these interrupts
- interrupt pending bits for all the interrupts
- interrupt enable bits for all the interrupts
- use of correct mtvec or mtvecx (target address and vectoring mode)
- impact on exception cause

MikeOpenHWGroup, 2020-04-26:
Hi @Silabs-ArjanB, these are all good points.  The place to record these is in the [Verification Plan](https://github.com/openhwgroup/core-v-docs/tree/master/verif/CV32E40P/VerificationPlan).  At this point we do not have anyone allocated to verify this Feature and getting one or two Active Contributors started on this is a top priority for me.

closed 2020-06-17 by davideschiavone