openhwfoundation/programs #35: Non-Maskable Interrupt

Imperas, 2020-05-01:
Looking at the RTL I see a signal at the top level for irq_nmi_i but I do not see any description in the manual for NMI, in particular when asserted does the core react to this interrupt, and if so what is the vector address for NMI ?
thx
Lee

Imperas, 2020-05-01:
@davideschiavone @MikeOpenHWGroup 
I also think there is a bug with the implementation of the NMI interrupt

The interrupt is only taken if the MSTATUS[MIE] is enabled, this is not correct as NMI is by definition 'non-maskable'
From the privilege spec

> Non-maskable interrupts (NMIs) are only used for hardware error conditions, and cause an immediate jump to an implementation-defined NMI vector running in M-mode **regardless of the state of a hart’s interrupt enable bits.**

Thx
Lee

davideschiavone, 2020-05-01:
That's correct you are right. This should be as a separate issue that this one on the core IP repository (as it is an HW bug).

Thanks
Davide

Silabs-ArjanB, 2020-06-16:
@Imperas The NMI pin has been removed from CV32E40P (see https://github.com/openhwgroup/core-v-docs/pull/56 and https://github.com/openhwgroup/core-v-docs/issues/50). Could you please close this issue?

closed 2020-06-17 by davideschiavone