openhwfoundation/programs #40: Table formatting

silabs-PaulZ, 2020-05-07:
The table formatting in the Verification Strategy (see https://core-v-docs-verif-strat.readthedocs.io/en/latest/sim_tests.html#virtual-peripherals) is hard to read. The content does not wrap and requires using the scroll bar. The PDF is also not formatted cleanly.

This formatting may possibly be fixed by overriding the theme, as seen in Ibex's conf.py

```
html_static_path = ['_static']

html_context = {
    'css_files' : [
        '_static/theme_overrides.css', # Fix wide tables in RTD theme
        ],
    }
```

jm4rtin, 2020-05-20:
I think that comes from: https://rackerlabs.github.io/docs-rackspace/tools/rtd-tables.html

When you turn this on (for the CV32E40P User Guide) the instruction encoding tables start wrapping all the cells:
[image]

So the solution may not be quite so simple.

MikeOpenHWGroup, 2020-11-15:
Seems to work for the tables in the VerifStrat, so I pushed it on top of PR #262, so poor @Silabs-ArjanB get to review it.   Thanks for the tip @silabs-PaulZ.

MikeOpenHWGroup, 2020-11-15:
Marked as resolved.   @silabs-PaulZ can you review and close (once the merge gets done).

MikeOpenHWGroup, 2020-11-17:
Hi @silabs-PaulZ, can you have an look?  (Yes, I'm in "nag" mode these days...)

silabs-PaulZ, 2020-11-17:
looks good, thanks

closed 2020-11-17 by silabs-PaulZ