openhwfoundation/programs #48: Pinout: Document Auxiliary Processing Unit (APU) interface

Silabs-ArjanB, 2020-05-18:
Currently there is no documentation about this interface in the User Manual. I added the core-v-docs/cores/cv32e40p/user_manual/source/apu.rst file to start the documentation for this feature including a list of the associated pins (and will do a pull request for this soon). 

We need to have (at least) the following information in this section:

- Description of the mentioned signals
- Protocol (and semantics) of the mentioned signals
- High level purpose of this interface
- Clarification of the WAPUTYPE value (it is currently 0 leading to apu_master_type_o[-1:0] (negative range)
- Interaction with the FPU
- Relationship with the FPU parameter (if any)
- Relationship with the PULP_ZFINX parameter (if any)

MikeOpenHWGroup, 2020-07-27:
@davideschiavone, @Silabs-ArjanB, I am not sure why this issue was closed.  Very few of the goals defined above have been met.

Silabs-ArjanB, 2020-07-28:
Hi @MikeOpenHWGroup As you can see this issue has not/never been closed. Only the duplicate issues https://github.com/openhwgroup/cv32e40p/issues/49 and #68 have been closed.

MikeOpenHWGroup, 2020-07-28:
Ha!  Right you are @Silabs-ArjanB.  I saw the "closed" icons in the discussion thread and incorrectly thought they applied to _this_ issue, not #49.  My mistake.

MikeOpenHWGroup, 2020-10-26:
@davideschiavone, do you have plans to do this documentation?  It is currently gating openhwgroup/cv32e40p#537.

davideschiavone, 2020-10-26:
Working on it with ETH, anyway that issue is a low priority task, not gating any verification or design issues

MikeOpenHWGroup, 2020-10-26:
>  issue is a low priority task, not gating any verification or design issues

I agree that it low priority, but we are pushing to RTL Freeze, so we either need to resolve all open issues or waive them.

davideschiavone, 2020-10-27:
#233 has been merged. Closing this issue.

MikeOpenHWGroup, 2020-10-27:
I do not think we can yet close this issue.  The current head of the docs (f2355b88) falls short of the goals of this issue.

MikeOpenHWGroup, 2020-11-03:
This issue has a PARAM:FPU label which means that it will be waived for Functional RTL Freeze.   I cannot agree with this since the APU interface of the core we are freezing does not match the ports specified in the Use Manual.  We may be able to defer full and complete documentation of this interface, but at least the ports on the RTL and the UM should match.

Silabs-ArjanB, 2020-11-03:
Hi @MikeOpenHWGroup I fixed the apu_* names in the integration chapter (see https://github.com/openhwgroup/core-v-docs/pull/243). As far as I am concerned we are then just waiting on completion of https://github.com/openhwgroup/core-v-docs/pull/237, which does not have to gate RTL freeze as far as I am concerned.

MikeOpenHWGroup, 2020-11-03:
Yep, I saw #243, thanks.  I agree that #237 does not gate RTL Freeze.   But we have time, so why not?