openhwfoundation/programs #50: Pinout: Document IRQ interface

Silabs-ArjanB, 2020-05-18:
As part of finishing the interrupt (and exception) related documentation I propose that we do the following:

- Remove NMI functionality (see below for the reasoning)
- Unify irq_fast_i and irq_fastx_i (see below for the reasoning)
- Make irq_id_o[5:0] and irq_ack_o behave the same for all interrupts (see below for the reasoning)

**NMI**

The proposal is to remove the NMI (irq_nmi_i) functinoality from the RTL (and replace it by a regular fast interrupt (irq_fast_i[15]). There are several reasons for this:

- Existing bugs, e.g. https://github.com/openhwgroup/cv32e40p/issues/336 and https://github.com/openhwgroup/cv32e40p/issues/323
- NMI will be standardized in the official RISC-V specification and will surely be non-compatible with our implementation (inherited from Ibex)
- Ibex implemented additional (non-standard) CSRs for backing up mstatus.MPP, mstatus.MPIE and mcause, whereas we do not have these features.

**Unify irq_fast_i and irq_fastx_i**

Proposal is to unify irq_fast_i[14:0],  and irq_fastx_i[31:0] into  irq_fast_i[47:0] (also replacing NMI by irq_fast_i[15] as described above), to rename MIEX/MIPX to MIE1/MIP1 respectively, and to replace MTVECX by MTVEC. The reasons are as follows:

- The user does not care about the differentiation between irq_fast_i and irq_fastx_i (they are all 'fast interrupts')
- For the same reason all interrupts should use MTVEC as their vector base address (the irq_fastx_i should not be treated as special with their own MTVECX base register).
- The current MTVECX base register adds to the (user perceived) complexity and does not really scale when we would add other modes in future chips; we would get MTBVEC, MTVECX, UTVEC, UTVECX, etc.)
- Unifying them allows for straight forward width increase in future projects.
- Renaming MIEX/MIPX to MIE1/MIP1 allows for straight forward fast interrupt addition in future projects (we would just add MIE2/MIP2, etc., not MIEXX/MIPXX)

**irq_id_o and irq_ack_o**

irq_id_o and irq_ack_o currently do not apply to all interrupts, i.e. they are not used for irq_fastx_i[31:0]. As indicated the idea is to treat all interrupts in the same manner and the proposal therefore is to extend irq_id_o by 1 MSB and also signal irq_ack_o for what are now called the irq_fastx_i. Issue was first raised in https://github.com/openhwgroup/cv32e40p/issues/243

Silabs-ArjanB, 2020-05-18:
Agreement was reached to accept the above proposals related to:

- Remove NMI functionality (see below for the reasoning)
- Make irq_id_o[5:0] and irq_ack_o behave the same for all interrupts (see below for the reasoning)

We are awaiting more opinions on the 'unify irq_fast_i and irq_fastx_i' topic.

davideschiavone, 2020-05-19:
Hi Arjan (@Silabs-ArjanB )

I had a meeting with Germain (@haugoug) and we thought that:

1.  MTVECx can be removed and every interrupts can be aligned to MTVEC but...

2.  To avoid creating holes in the Interrupt Vector Table (IVT), it would be better to implement also the "holes" interrupts like:

- User software interrupt
- Supervisor software interrupt
- Reserved for future standard use (custom with id 2)
- User timer interrupt
- Supervisor timer interrupt
- Reserved for future standard use (custom with id 6)
- User external interrupt
- Supervisor external interrupt
- Reserved for future standard use (custom with id10)

All interrupts should be taken in machine mode and their usage should be as the other ones

Silabs-ArjanB, 2020-05-26:
Hi @davideschiavone and  @haugoug,

Paul and I discussed your proposal and we think the holes that are reserved by the RISC-V spec should not be used by our custom extensions.

I have a counter proposal for how we can remove MTVECX CSR and still have the compact vector table that you are looking for. The proposal is based on a local parameter IGNORE_CAUSE_MSB that you can change to map what used to be called irq_fastx_i[:0] to be mapped to the MTVEC + 0.

The main related RTL change (other than actually removing MTVECX) would mainly be the following:

```
module riscv_if_stage:
(
    input  logic  [**5**:0] exc_vec_pc_mux_i,      // selects ISR address for vectorized interrupt lines
...

  **localparam IGNORE_CAUSE_MSB = 0**; // Ignore the MSB of the exception code (effectively mapping the top 32 and bottom 32 IRQs on top of each other)

  // exception PC selection mux
  always_comb
  begin : EXC_PC_MUX
    exc_pc = '0;

    unique case (trap_addr_mux_i)
      TRAP_MACHINE:  trap_base_addr = m_trap_base_addr_i;
      TRAP_USER:     trap_base_addr = u_trap_base_addr_i;
      // All references to TRAP_MACHINEX will be replaced by TRAP_MACHINE and 
      // m_trap_base_addr_i will be used instead of m_trap_base_addrx_i
      **// TRAP_MACHINEX:** trap_base_addr = m_trap_base_addrx_i; 
       default:;
    endcase

    unique case (exc_pc_mux_i)
      EXC_PC_EXCEPTION:                        exc_pc = { trap_base_addr, 8'h0 }; //1.10 all the exceptions go to base address
      EXC_PC_IRQ:                              exc_pc = { trap_base_addr, **IGNORE_CAUSE_MSB ? {1'b0, exc_vec_pc_mux_i[4:0]} : exc_vec_pc_mux_i[5:0]**, 2'b0 }; // interrupts are vectored
      EXC_PC_DBD:                              exc_pc = { DM_HALTADDRESS       };
      default:;
    endcase
  end

```

davideschiavone, 2020-05-27:
that looks very elegant to me, so by tieing to 0 the first "32" interrupts, one have the last 32 still jumping at the beginning of the IVT. Which is what would make the core like the one for PULP today.

I personally agree, let's wait for @haugoug

Thanks a lot
Davide

haugoug, 2020-05-29:
Hi @Silabs-ArjanB,
thanks a lot. I still think the solution of having 32 symmetrical interrupts is more elegant, but I agree your proposal is a good compromise at short term, let's proceed with it.

Silabs-ArjanB, 2020-05-29:
Hi @haugoug , thanks for the feedback. I will start working on the actual RTL changes.

Silabs-ArjanB, 2020-06-09:
@davideschiavone , @haugoug We again discussed Davide's proposal above and we now think that it is a good idea if we extend Davide's proposal with a parameter to determine which of the 64 interrupt lines are actually implemented. In that way the integrator can still chose to not use the reserved indices, but it will also be possible to 'fill the holes'. I will write up a more detailed proposal shortly.

Silabs-ArjanB, 2020-06-09:
Here is the proposal:

A) Merge the existing irq_*_i signals into one irq_i vector, so:

  Old:

    input  logic        irq_software_i,
    input  logic        irq_timer_i,
    input  logic        irq_external_i,
    input  logic [47:0] irq_fast_i,
    input  logic        irq_nmi_i,

  New:

    input  logic [63:0] irq_i,

B) Control the irq_i[] lines by two new parameters: 

    parameter IRQ_MASK        = 64'h0000_0000_0000_0888, // IRQ implementation mask
    parameter IRQ_PRIO_SCHEME = 0,                       // 0 = RISC-V compliant scheme, 1 = custom scheme

There will be two interrupt schemes: 'RISC-V compliant' and 'custom'.

B0) RISC-V compliant scheme

  - irq[31:16] are optional (corresponding IRQ_MASK bits are user defined)
  - irq[11] corresponds to MEI (Machine External Interrupt) and shall not be masked out (corresponding IRQ_MASK bit shall be 1)
  - irq[ 7] corresponds to MTI (Machine Timer Interrupt)    and shall not be masked out (corresponding IRQ_MASK bit shall be 1)
  - irq[ 3] corresponds to MSI (Machine Software Interrupt) and shall not be masked out (corresponding IRQ_MASK bit shall be 1)

  - irq 63-32 shall be be masked out (corresponding IRQ_MASK bits shall be 0)
  - 15-12, 9-8, 6-4, 2-0 shall be masked out (corresponding IRQ_MASK bits shall be 0)

  - Priority assignments are RISC-V compliant (from highest to lowest):

    irq[31]
    irq[30]
    ...
    irq[16]
    irq[11]  MEI
    irq[ 3]  MSI
    irq[ 7]  MTI

  On future CV* cores the not implemented irqs in the 15-0 range might get defined in the RISC-V specification (at a to-be-decided priority level).

  MIP1 and MIE1 will not be present (illegal instruction upon access).

B1) Custom scheme

  - irq[63:0] are optional (corresponding IRQ_MASK bits are user defined)

  - Priority assignments are NOT RISC-V compliant, but easy to explain (monotonous priorities) (from highest to lowest):

  irq[63]
  irq[62]
  ...
  irq[0]

  - irq[63:32] use the MIP1 and MIE1 CSRs (these CSRs are always present (but might appear as hard wired to 0 depending on IRQ_MASK))
  - irq[31:0] use the MIP and MIE CSRs (these CSRs are always present (but might appear as hard wired to 0 depending on IRQ_MASK))

For both interrupt priority schemes MTVEC is used as the trap vector base address (and the irq index is used for vectoring).

MikeOpenHWGroup, 2020-06-09:
Option B seems more complex from a verification perspective, but it seems to be a path towards a unified interrupt strategy for all CORE-V cores (which is less complex in the long run).

Both A and B are better than what we have now.

Have we reached out to em-micro about this (probably not, since they just signed-on as OpenHW members).  @GTumbush what do you think?

Silabs-ArjanB, 2020-06-09:
@MikeOpenHWGroup Just to be sure that the above is not misunderstood. In above proposal A) and B) are not 'options' of which we would implement only 1; the proposal is to implement both and at compile time the user makes the choice via a parameter.

davideschiavone, 2020-06-09:
Hi everyone, 

I am also in favour for option A) for simplicity but, the priority order would still exist as int[3] has higher priority than [7] for compliancy.

How would we deal with this? 

Probably option B) works better as by default, the core will be compliant with only a subset of interrupts being implemented. 

I would probably go for A. but defining a CUSTOM priority (with no parameters at all)

This priority can be 

irq[63]
...
irq[32]
irq[31]
irq[30]
...
irq[16]
irq[15]
...
irq[12]
irq[11] MEI
irq[ 3] MSI
irq[ 7] MTI
irq[10]
irq[9]
irq[8]
irq[2]
irq[1]
irq[0]
irq[6]
irq[5]
irq[4]

I know it's strange, but if we document it properly it should not be a problem.
In this way, users that do not want to use custom interrupts, will simply tie to 0 the interrupts line that are not used, and priority is thus not affected.
Who wants to use 32 or 64 interrupts, should be aware of this order scheme and if they don't like it, they should simply change the way they wire interrupts from outside.

What do you think?

jm4rtin, 2020-06-09:
What we currently use is a (mostly) CLIC compatible modification of CV32E40P. We have essentially disabled CLINT and tie the following signals to 0:
* irq_software_i
* irq_timer_i
* irq_external_i
* irq_fast_i

We also removed fastx (and NMI prior to your change) as it is in the way of the CLIC interrupts that start at ID 16. We went back to an interface similar to what the core used to have:
```sv
input  logic                 irq_i,
input  logic [IRQ_ID_W-1:0]  irq_id_i,
input  logic [IRQ_LEV_W-1:0] irq_lev_i,
output logic                 irq_ack_o,
output logic [IRQ_ID_W-1:0]  irq_id_o,
```

We then implement the memory-mapped portion of the CLIC specification as an external block. This block does the prioritization of the next interrupt to be handled. This leaves the core only worrying about when an interrupt handler should be vectored to (based on mstatus.xie, the current privilege level and the current interrupt priority level). Arbitration of interrupts in this setup happens outside the core. The one thing we don't handle right now is horizontal interrupts.

It is a good step in the right direction; however, I don't think either solution addresses CLIC or future-proofs and unifies the interrupt interface across cores. I would rather not have a 1024-bit vector going into the core when CV32E40 starts adding CLIC. If you don't want to change the interrupt system again in the future it might be a good idea to consider the implications of implementing CLIC.

Silabs-ArjanB, 2020-06-10:
@jm4rtin Fully agreed that we need to take CLIC into account as well. I would like a proposal that satisfies the following constraints:

- Ability to be CLINT compatible (one implementation would be the above proposal with IRQ_PRIO_SCHEME = 0)
- Ability to use the 'holes in the CLINT vector table' with a priority scheme that is monotonous (so not CLINT compatible) (one implementation would be the above proposal with IRQ_PRIO_SCHEME = 1)
- Future proof with CLIC (ideally by not changing the pinout)

I will write up a more detailed proposal taking your interface proposal into account as well.

Silabs-ArjanB, 2020-06-16:
Note: This discussion is continued in https://mattermost.openhwgroup.org/all-users/channels/twg--cores--interrupts

Silabs-ArjanB, 2020-06-17:
Decided to stick with the original implementation (before the June 9 remarks). Only renamed some pins to make the interface easier to understand and future proof for CLINT based cores that might add User mode.

See https://github.com/openhwgroup/core-v-docs/pull/100 for details.

closed 2020-06-17 by Silabs-ArjanB