openhwfoundation/programs #51: Module: Rename riscv_core to cv32e40p _core

Silabs-ArjanB, 2020-05-18:
The currently used module names (and file names) are not future proof. They are a bit generic and will lead to Verilog naming conflicts once we introduce multiple CV* cores. Proposal is to adopt naming similar to as is done in the Ibex projects (where each module name is prefixed with ibex_).

This task is to rename the the module name from riscv_core to cv32e40p _core and the associated filename from riscv_core.sv to cv32e40p _core.sv.

(All other files/modules need to be renamed as well, but that will be done as a separate step as it does not impact the documentation and verification in the same manner).

MikeOpenHWGroup, 2020-05-18:
This is going to be a minor pain, but I strongly support this proposal.   We should also pre-pend the OpenHW "solder-pad" license header (leaving the existing header in place) and add compile-guards:
```
`ifndef __CV32E40P_ALU__
`define __CV32E40P_ALU__
import riscv_defines::*;

module riscv_alu
....
endmodule
`endif // __CV32E40P_ALU__

Silabs-ArjanB, 2020-05-19:
Fix ready on local branch. Waiting with pull request issue until other pull requests mentioned in https://github.com/openhwgroup/core-v-docs/issues/45 have been executed.

Silabs-ArjanB, 2020-05-19:
Still need to discuss the additional topics raised above (ifdefs, license header). Proposing to create separate topics for these in the CV32E40P repos as they do not really touch on the documentation.

Silabs-ArjanB, 2020-05-27:
Discussed Mike's comments above with Mike, Paul, Davide. Agreed to not add the compile guards (`ifndef, etc.). The license header topic will be tracked via a separate issue (https://github.com/openhwgroup/cv32e40p/issues/355).

closed 2020-06-16 by Silabs-ArjanB