openhwfoundation/programs #58: Pinout: Document usage of fregfile_disable_i

Silabs-ArjanB, 2020-05-18:
The usage of the fregfile_disable_i is not documented and not clear. It might not even be needed anymore.

A relevant part of the code related to this is the following from https://github.com/openhwgroup/cv32e40p/blob/master/rtl/riscv_id_stage.sv:

  assign fregfile_ena = FPU && !PULP_ZFINX ? ~fregfile_disable_i : '0;

Task is to figure out how this pin is supposed to be used and then to either document it or remove it (if the use case is no longer valid).

davideschiavone, 2020-05-19:
Dear @Silabs-ArjanB ,

I had a call with @haugoug, SW leader of the PULP team.

This pin was indeed only for exploration on prototypes. We can remove it.

best
Davide

Silabs-ArjanB, 2020-05-19:
Perfect; I will update the documentation and I will issue a pull request for the removal of this pin in the RTL.

Silabs-ArjanB, 2020-05-19:
Fixes in cv32e40p repos ready on local branch; waiting for pull requests mentioned in #45 before issuing this one.

davideschiavone, 2020-05-25:
Shall we close this issue as the pin has been removed?

Silabs-ArjanB, 2020-05-25:
Pin no longer present

closed 2020-05-25 by Silabs-ArjanB