openhwfoundation/programs #62: Pinout: Add dm_exception_addr_i pin

silabs-PaulZ, 2020-05-26:
The proposal is to add a dm_exception_addr_i[31:0] pin.

This will provide an address to jump to when an exception occurs when executing code during debug mode.

The reasons for this addition are as follows:

- We want to use a pin as opposed to a parameter to match how boot_addr_i and dm_halt_addr_i is implemented
- This will address the issue https://github.com/openhwgroup/cv32e40p/issues/185, where exceptions are not handled correctly during debug mode
- This solution is the same as seen on the IBEX core

Silabs-ArjanB, 2020-05-26:
Must this be word-aligned (just like the dm_halt_addr_i)?

silabs-PaulZ, 2020-05-26:
Yes, the dm_halt_addr_i needs to be word-aligned.

MikeOpenHWGroup, 2020-05-26:
Sounds like a great thing to specify in the user_manual.   Sorry, couldn't resist.  :wink:

silabs-PaulZ, 2020-05-26:
I decided to make this word aligned (in the documentation pull request) to ease verification and software environment concerns. We can change later if there is a compelling advantage for non-word aligned debug entry points.

Silabs-ArjanB, 2020-05-27:
> 
> 
> Sounds like a great thing to specify in the user_manual. Sorry, couldn't resist. 😉

This is already in the documentation and awaiting its pull request. I was just checking that the documentation update was correct.

silabs-PaulZ, 2020-06-08:
Implemented.

closed 2020-06-08 by silabs-PaulZ