openhwfoundation/programs #75: Typo: RI5CY User Manual v4.2, description of p.add* instructions

alfredoh1234, 2020-06-04:
@davideschiavone , in RI5CY user manual:
https://github.com/pulp-platform/riscv/blob/master/doc/user_manual.doc

* In section 14.3.3 General ALU Operations, the description of p.addN has following note:
  - Note: Arithmetic shift right. Setting Is3 to 2 replaces former p.avg
* In section 14.3.3 General ALU Operations, the description of p.adduN has following note:
  - Note: Logical shift right. Setting Is3 to 2 replaces former p.avg
* In section 14.3.3 General ALU Operations, the description of p.addRN has following note:
  - Note: Arithmetic shift right.
* In section 14.3.3 General ALU Operations, the description of p.adduRN has following note:
  - Note: Logical shift right.
* In section 14.3.3 General ALU Operations, the description of p.addNr has following note:
  - Note: Arithmetic shift right. 
* In section 14.3.3 General ALU Operations, the description of p.adduNr has following note:
  - Note: Logical shift right. 
* In section 14.3.3 General ALU Operations, the description of p.addRNr has following note:
  - Note: Arithmetic shift right.
* In section 14.3.3 General ALU Operations, the description of p.adduRNr has following note:
  - Note: Logical shift right.

It seems that ls3 should be replaced by 2^(ls3-1) like this : 

 **p.addN** | |  rD = (rs1 + rs2) >>> 2^(Is3-1)
 |--|--|--|
 **p.adduN** | |  rD = (rs1 + rs2) >> 2^(Is3-1)
 **p.addRN** | |  rD = (rs1 + rs2 + 2^(Is3-1)) >>> 2^(Is3-1)
 **p.adduRN** | |  rD = (rs1 + rs2 + 2^(Is3-1)) >> 2^(Is3-1)

* In section 14.3.3 General ALU Operations, the description of p.addNr has following note:
  - Note: Arithmetic shift right. 
* In section 14.3.3 General ALU Operations, the description of p.adduNr has following note:
  - Note: Logical shift right. 
* In section 14.3.3 General ALU Operations, the description of p.addRNr has following note:
  - Note: Arithmetic shift right.
* In section 14.3.3 General ALU Operations, the description of p.adduRNr has following note:
  - Note: Logical shift right.

It seems that rs2[4:0] should be replaced by 2^(rs2[4:0]-1) like this : 

 **p.addNr** | |  rD = (rD + rs1) >>> 2^(rs2[4:0]-1)
 |--|--|--|
 **p.adduNr** | |  rD = (rD + rs1) >> 2^(rs2[4:0]-1)
 **p.addRNr** | |  rD = (rD + rs1 + 2^(rs2[4:0]-1)) >>> 2^(rs2[4:0]-1)
 **p.adduRNr** | |  rD = (rD + rs1 + 2^(rs2[4:0]-1)) >> 2^(rs2[4:0]-1)

***
Alfredo

davideschiavone, 2020-06-04:
Hello @alfredoh1234 , 

Why do you think the imm field should be replaced by 2^(imm-1) ?

It is actually correct like this. The RTL does what it's written in the doc right now
https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html

davideschiavone, 2020-06-15:
Hi @alfredoh1234 , any comment on it?

alfredoh1234, 2020-06-16:
@davideschiavone , my mistake. when I saw rs2[4:0] I thought at the time it was 4-bits and not 5-bits, thus limiting the range of the shift.
You can close this issue without fixing anything. Sorry about that

closed 2020-06-17 by davideschiavone