openhwfoundation/programs #76: Read the Docs "Control and Status Registers" table has errors

alfredoh1234, 2020-06-04:
The rendering of the  "[Control and Status Registers](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html)" table in Read the Docs and its [source file in the repository ](https://github.com/openhwgroup/core-v-docs/blob/master/cores/cv32e40p/user_manual/source/control_status_registers.rst) have errors

* Memory range assigned twice:

CSR Address | Hex | Name | Acc. | Description
-- | -- | -- | -- | --
01 | 11 | 10 | 110xxx | 0x7B0-0x7B7 | HWLP | R/W | Hardware Loop Registers
-- | -- | -- | -- | --
01 | 11 | 10 | 110000 | 0x7B0 | DCSR | R/W | Debug Control and Status
01 | 11 | 10 | 110001 | 0x7B1 | DPC | R/W | Debug PC
01 | 11 | 10 | 110010 | 0x7B2 | DSCRATCH0 | R/W | Debug Scratch Register 0
01 | 11 | 10 | 110011 | 0x7B3 | DSCRATCH1 | R/W | Debug Scratch Register
-- | -- | -- | -- | --

davideschiavone, 2020-06-08:
This should be fixed in https://github.com/openhwgroup/core-v-docs/pull/91

davideschiavone, 2020-06-15:
Is that ok?

alfredoh1234, 2020-06-15:
From the other thread, it appears that the HWLoop is only implemneted when "if the PULP_HWLP parameter is 1".
Does it mean that when the PULP_HWLP is set to 1, the Debug registers and the HW-Loop registers are shared? 
Please clarify

davideschiavone, 2020-06-15:
No why do you say so? The HWLoop are in 7C*, the debug registers in 7B*

alfredoh1234, 2020-06-16:
I see that the table was updated and this issue can now be closed

-- 

***
Alfredo,
IEEE senior member,
IEEE-SA Open Source Committee (OSCom) member,
IEEE-Ottawa SIGHT chair.
~~~https://www.linkedin.com/in/alfredoherreraeng
skype: alfredoh1234
​twitter: @alfredoh1234​
~*~*~*
IEEE. Advancing Technology for Humanity.

closed 2020-06-17 by davideschiavone