openhwfoundation/programs #80: RI5CY: User Manual v4.2-14.5.1 Vectorial ALU Operations table missing information

alfredoh1234, 2020-06-05:
The Description for the Mnemonics is missing information to understand how the instructions operate. 
For example, the indexing variable "i" range and meaning are not defined; pv.*.sc.* operand use is not clearly described; carry/borrow handling are not described; and the use of the "0xFFFF" mask is undefined in all cases.
For example:
* pv.add.h is understood as:
  * rD[31-16] = rs1[31-16] + rs2[31-16] 
  * rD[15-0] = rs1[15-0] + rs2[15-0]
* Is pv.add.sc.h understood as?
  * rD[31-16] = rs1[31-16] + Zext(rs2[15-0])
  * rD[15-0] = rs1[15-0] + Zext(rs2[15-0])
* Is pv.add.sci.h understood as?
  * rD[31-16] = rs1[31-16] + Zext(Imm6[5-0])
  * rD[15-0] = rs1[15-0] + Zext(Imm6[5-0])

davideschiavone, 2020-06-05:
Dear @alfredoh1234 ,

here there is a fix that hopefully cover your suggestion.

https://github.com/davideschiavone/core-v-docs/commit/a67b42a1d607dd2135747fedf3b2c2ada4817ea7

Is that ok? 
thanks

alfredoh1234, 2020-06-05:
I think it is better, thanks!

closed 2020-06-08 by davideschiavone