openhwfoundation/programs #81: RI5CY: User Manual v4.2-14.5.1 pv.*.div2, pv.*.div4 and pv.*.div8 unclear

alfredoh1234, 2020-06-05:
The description for the pv.*.div2, pv.*.div4 and pv.*.div8 instructions cannot be understood from description.
Current descriptions do use the *.h appendix to indicate operations are on 16-bit operands. Also, rs2 is refereed to in section 14.5.2 but not in section 14.5.1. Finally, it is unclear why the "0xFFFF" mask may be needed and where it is stored to be used (is it hard-coded?)

For example:
* pv.add.div2 rD, rs1, rs2
  * rD[0] = (rs1[0] + rs2[0])/2
  * rD[1] = (rs1[1] + rs2[1])/2
* pv.add.div4 rD, rs1, rs2
  * rD[0] = (rs1[0] + rs2[0])/4
  * rD[1] = (rs1[1] + rs2[1])/4
* pv.add.div8 rD, rs1, rs2
  * rD[0] = (rs1[0] + rs2[0])/8
  * rD[1] = (rs1[1] + rs2[1])/8

davideschiavone, 2020-06-05:
Hi @alfredoh1234 ,

the issue #80 solves the 0xFFFF mask information. Whereas the rs2 operator has been fixed here
https://github.com/davideschiavone/core-v-docs/tree/fix81

Is that ok?

best
Davide

alfredoh1234, 2020-06-11:
@davideschiavone , I cannot tell from the link you shared (the new User Manual does not have any chapter/section numbering or pages :-( ). 

If I found the right section on the new user manual, this seems much better:
   pv.add{.div2,.div4, .div8} | rD[i] = ((rs1[i] + op2[i]) & 0xFFFF)>>{1,2,3}

I think you can lose this issue

closed 2020-06-15 by davideschiavone