openhwfoundation/programs #82: RI5CY: User Manual v4.2-14.5.1 pv.subrotmj.* instruction undefined

alfredoh1234, 2020-06-05:
There is no description of what the pv.subrotmj.* instruction is. From another document on the web, it seems like it is the "substraction of two 16-bit complex numbers with post rotation by -j" or
* rD[15:0 ] =   rA[31:16] - rB[31:16]
* rD[31:16] = -(rA[15:0 ] - rB[15:0])

Please clarify

alfredoh1234, 2020-06-05:
Because when writing the verification code it is important to know the expected result, boundary conditions and how to insert cases that will create an expected error to test any exception handling or flags in the CSRs

davideschiavone, 2020-06-08:
The fix is now in 
https://github.com/openhwgroup/core-v-docs/pull/87

Can you please confirm and give me feedback?

thanks
Davide

davideschiavone, 2020-06-16:
@alfredoh1234 can you please add comment?

alfredoh1234, 2020-06-16:
It is better. You can close [issue #82](https://github.com/openhwgroup/core-v-docs/issues/82)

PR #87 (2020-06-17): complex instructions now in a separate sub section
Separated the complex instructions from the general SIMD instructions
Added more info about the meaning of such instructions
Fixed #82 and #86
closed 2020-06-17 by davideschiavone