openhwfoundation/programs #86: RI5CY: User Manual v4.2-14.5.1 error in complex imaginary multiplication pv.cplxmul.i.*

alfredoh1234, 2020-06-08:
If using complex convention where rD[15-0] is the real part and rD[31-16 is the imaginary part:
* pv.cplxmul.i.{/,div2,div4,div8} it wrong
  * **Current:** rD[ **15:0** ] = (rs1[0]\*rs2[1]   + rs1[1]\*rs2[0])>>{15,16,17,18}   and rD[ **31:16** ] = rD[ **31:16** ]
  * **Should be?:** rD[ **31:16** ] = (rs1[0]\*rs2[1]   + rs1[1]\*rs2[0])>>{15,16,17,18}   and rD[ **15:0** ] = rD[ **15:16** ]

davideschiavone, 2020-06-08:
The fix should be now in https://github.com/openhwgroup/core-v-docs/pull/87

davideschiavone, 2020-06-16:
@alfredoh1234 can you please add comment?

alfredoh1234, 2020-06-16:
The text is clearer, you can close [issue #86](https://github.com/openhwgroup/core-v-docs/issues/86)

closed 2020-06-17 by davideschiavone