openhwfoundation/programs #88: RI5CY: User Manual v4.2-14.5.1 missing information for pv.shuffle2.* instruction

alfredoh1234, 2020-06-08:
The pv.shuffle.h and pv shuffle2.b instructions assign parts of rs1 to rD, but the instruction description doesn't indicate which part:
For example:
* pv.shuffle2.h:
  * rD[31:16] = ((rs2[17] == 1) ? rs1 **[missing information here]**
* pv.shuffle2.b:
  * rD[31:24] = ((rs2[26] == 1) ? rs1 **[missing information here]**

davideschiavone, 2020-06-08:
Hi @alfredoh1234 , I think it does. It is written in the documentation.
It uses the rs2 register to address which part.
Can you please confirm it?
thanks

davideschiavone, 2020-06-15:
Hi @alfredoh1234  , any comment on it?

alfredoh1234, 2020-06-16:
I am still not certain what this code means:
* rD[31:24] = ((rs2[26] == 1) ? rs1 :   rD)[rs2[25:24]*8+7:rs2[25:24]*8]   
* rD[23:16] = ((rs2[18] == 1) ? rs1 : rD)[rs2[17:16]*8+7:rs2[17:16]*8]  
* rD[15:8] = ((rs2[10] == 1) ? rs1 : rD)[rs2[9:8]*8+7:rs2[9:8]*8] 
*  rD[7:0] = ((rs2[2] == 1) ? rs1 : rD)[rs2[1:0]*8+7:rs2[1:0]*8]

1. Is the first parenthesis after the equal signing an if statement selecting either rs1 or rD based on the value of one bit for each byte?
2. The bracket after the parenthesis seems to indicate the bits from rs2 used to select from either rs1 or rD

Is this correct. If so, then you can close the issue. If not, please clarify.

davideschiavone, 2020-06-17:
Hi @alfredoh1234 , 

rs2[26] == 1 says whether the register rs1  or rD are used
[rs2[25:24]*8+7:rs2[25:24]*8] indicate which parts of rs1 or rD are used

So the first bit select which register, the second set of bits select which part of that register

Is that clearer?

alfredoh1234, 2020-06-17:
Yes, please close the issue

-- 

***
Alfredo,
IEEE senior member,
IEEE-SA Open Source Committee (OSCom) member,
IEEE-Ottawa SIGHT chair.
~~~https://www.linkedin.com/in/alfredoherreraeng
skype: alfredoh1234
​twitter: @alfredoh1234​
~*~*~*
IEEE. Advancing Technology for Humanity.

closed 2020-06-18 by davideschiavone