openhwfoundation/programs #96: Bit width discrepancy in multiply-accumulate documentation

DavidM-EMUS, 2020-06-11:
In the documentation at the below link:
 https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html#multiply-accumulate

In section Pulp Instruction Set Extensions > Multiply-Accumulate, bit width for instructions that utilise the upper half of the input registers have the bit range used as [31:15], when they should state that the range is [31:16]. For example, for instruction p.muls, the description of its functionality reads as such:
rD[31:0] = Sext(rs1[15:0]) * Sext(rs2[15:0])

while instruction p.mulhhs, (its upper-half utilising counterpart) reads as such:
rD[31:0] = Sext(rs1[31:15]) * Sext(rs2[31:15])

This is faulty information for p.mulhhs, as testing with the rtl shows that the bit range used is [31:16]. This affects descriptions of 10 instructions in total:
p.mulhhs, p.mulhhsN, p.mulhhsRN, p.mulhhu, p.mulhhuN, p.mulhhuRN, p.machhsN, p.machhsRN, p.machhuN, and p.machhuRN

davideschiavone, 2020-06-16:
Hi @DavidM-EMUS , thanks for catching it!

I made a PR here: https://github.com/openhwgroup/core-v-docs/pull/98

Can you please provide feedback about it?

davideschiavone, 2020-06-16:
Reopen the issue as I am waiting for @DavidM-EMUS input

DavidM-EMUS, 2020-06-16:
Hi @davideschiavone The changes look good. Thanks for taking care of that!

commit 0b1acd3f (2020-06-16): Merge pull request #98 from davideschiavone/fix96

:bug: fix #96
commit 3903d057 (2020-06-16): :bug: fix #96
closed 2020-06-16 by DavidM-EMUS