openhwfoundation/programs #101: Multiply-Accumulate right shift notation incorrect

DavidM-EMUS, 2020-06-17:
In the document at the link below:
https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html#multiply-accumulate

In section Pulp Instruction Set Extensions > Multiply-Accumulate: For all unsigned instructions that also included a right shift, documentation specifies logical right shift, but uses three arrows instead of two. This is rather confusing and/or poorly worded as it could cause confusion as the mathematical equation denotes arithmetic right shifting. I know this is nitpicking, but the notes for instructions may be ignored by some users. Testing results show that RTL performs logical right shifting for these instructions. Example below was copied (with minor formatting for readability) from documentation at the link above:

p.muluN rD, rs1, rs2, Is3
rD[31:0] = (Zext(rs1[15:0]) * Zext(rs2[15:0])) >>> Is3 Note: Logical shift right

the second line of which could be simplified to:
rD[31:0] = (Zext(rs1[15:0]) * Zext(rs2[15:0])) >> ls3

Please note this effects a total of 8 instructions:
p.muluN
p.mulhhuN
p.muluRN
p.mulhhuRN
p.macuN
p.machhuN
p.macuRN
p.machhuRN

davideschiavone, 2020-06-22:
Thanks for reporting it, can you please check the PR #110 ?

DavidM-EMUS, 2020-06-22:
That fix looks good. Thank you for taking care of that!

davideschiavone, 2020-06-22:
Thanks a lot, closing once the merge happens.

PR #110 (2020-06-30): fix #101
This should fix #101
commit 606b58d2 (2020-06-30): Merge pull request #110 from davideschiavone/fix101

fix #101
commit 7e274712 (2020-06-30): :bug: fix #101
closed 2020-06-30 by davideschiavone