openhwfoundation/programs #128: Vplan: timing of debug_req_i pulse

MikeOpenHWGroup, 2020-07-10:
## Bug Title
Update [Debug Vplan](https://github.com/openhwgroup/core-v-docs/blob/master/verif/CV32E40P/VerificationPlan/debug-trace/CV32E40P_debug.xlsx) to consider timing of debug_req_i assertion and de-assertion.

### Type
Potentially missing information in the Vplan

### Decription of Issue
CV32E40P [issue #404](https://github.com/openhwgroup/cv32e40p/issues/404) highlights an issue with the timing of the assertion of debug_req_i.   To the best of my knowledge, we have no features captured in the Debug Vplan that would cause us to exercise and catch this and related cycle timing issues.   Please review the CV32E40P issue and make appropriate updates to the Debug Vplan.

### Additional context
Please discuss with @Silabs-ArjanB to see if there is a potential for similar issues.<br>
If there are similar issues related to features other than Debug, please bring them to my attention.

MikeOpenHWGroup, 2020-11-11:
Hi @silabs-oysteink, I am 99.9% certain you have already addressed this in the debug DV plan.   Please update this issue with a confirmation (or not).

silabs-oysteink, 2020-11-12:
There is no entry in the vplan to explicitly cover what's described in this issue. As far as I can see, the issue fixed in RTL, but I will add  some lines in the vplan to describe timing of debug_req_i.

MikeOpenHWGroup, 2020-11-12:
Excellent, thanks @silabs-oysteink.  When you do, please update this issue and I'll close it off.

MikeOpenHWGroup, 2020-11-17:
Hi @silabs-oysteink, so you have an update on this issue?

silabs-oysteink, 2020-11-17:
Updated vplan with info about 'debug_test_boot_set' which addresses this issue.

MikeOpenHWGroup, 2020-11-17:
Thanks!, I'll close this issue now.

closed 2020-11-17 by MikeOpenHWGroup