openhwfoundation/programs #130: CV32E40P User Manual: specification of privlv CSR

MikeOpenHWGroup, 2020-07-13:
### Type
Issues with this CSR as documented:
* `privlv` is not part of the RISC-V ISA so it needs to be in a "custom" address space.
* The CSR is located at 0xC10, which is "standard read-only" space.
* The "Current Priviledge Level" field contains lots of info that will never relate to CV32E40P.

### Location of Issue
1. RTD URL: https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html#privilege-level-privlv
2. core-v-docs Git Hash:  609007a

Silabs-ArjanB, 2020-07-13:
The https://github.com/openhwgroup/core-v-docs/pull/106 pull request was supposed to fix the CSR address part (but obviously the actual address was not updated). I will issue a new pull request to update the CSR address (same hold for UHARTID, PRIVLV, HWLOOP*, FPREC)

Silabs-ArjanB, 2020-07-13:
Justification for this custom CSR can be found in https://github.com/openhwgroup/cv32e40p/issues/381

PR #132 (2020-08-31): Correcting CSR addresses as mentioned in pull request #106
Updating CSR addresses as was originally intended in pull request #106 

```
CSR                     Old address     New address     Comment
---------------------------------------------------------------
HWLoop0_START           12'h7C0         12'h800         Needs to be User accessible (now in user custom read/write space)
HWLoop0_END             12'h7C1         12'h801         Needs to be User accessible (now in user custom read/write space)
HWLoop0_COUNTER         12'h7C2         12'h802         Needs to be User accessible (now in  [... cut]
closed 2020-08-31 by Silabs-ArjanB