openhwfoundation/programs #131: CV32E40P User Manual: specification of uhartid CSR

MikeOpenHWGroup, 2020-07-13:
### Type
Issues with this CSR as documented:
* `uhartid` is not part of the RISC-V ISA so it needs to be in a "custom" address space.
* `mhartid` **_is_** part of the RISC-V ISA and it is supported by CV32E40P.  How is this CSR different than that one?
* The CSR is located at 0x14, which is "standard read-only" space.
* The description of the CSR is incomplete (I can't tell what it does from the description).

### Location of Issue
1. RTD URL: https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html?highlight=privlv#user-hardware-thread-id-uhartid
2. core-v-docs Git Hash:  609007a

Silabs-ArjanB, 2020-07-13:
The https://github.com/openhwgroup/core-v-docs/pull/106 pull request was supposed to fix the CSR address part (but obviously the actual address was not updated). I will issue a new pull request to update the CSR address (same hold for UHARTID, PRIVLV, HWLOOP*, FPREC)

Silabs-ArjanB, 2020-07-13:
Justification for this custom CSR can be found in https://github.com/openhwgroup/cv32e40p/issues/381

Silabs-ArjanB, 2020-08-31:
Hi @MikeOpenHWGroup This issue should have been solved now with https://github.com/openhwgroup/core-v-docs/pull/178; can you please close it if you agree?

MikeOpenHWGroup, 2020-08-31:
Agreed!   Thanks.  Closing.

closed 2020-08-31 by MikeOpenHWGroup