openhwfoundation/programs #133: misa return value

MikeOpenHWGroup, 2020-07-13:
### Note
I'm not sure if this is an RTL bug or a documentation bug, but I've already filed a couple of documentation bugs, so I'll repeat that here.  If this should be an RTL, issue, I can re-create it there.

### Type
Issues with this CSR as documented:
* `misa` is documented to have a reset value of 0x0.
* Actual reset value is 0x40801104.
* This is actually what you'd expect from an RV32IMCX machine.
* Its not clear if CV32E40P should report support for non-standard instructions when PULP=0.

### Location of Issue
1. RTD URL: https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html?highlight=misa#machine-isa-misa
2. core-v-docs Git Hash:  609007a

Silabs-ArjanB, 2020-07-14:
This was both a documentation and RTL issue indeed. The documentation has been fixed in https://github.com/openhwgroup/core-v-docs/pull/132. The RTL has been fixed in https://github.com/openhwgroup/cv32e40p/pull/398.

Silabs-ArjanB, 2020-08-31:
Hi @MikeOpenHWGroup Can this one be closed now? (The corresponding PR has been merged)

MikeOpenHWGroup, 2020-08-31:
Almost, but not quite.  None of the fields of MISA are R/W.  They are all RO.

Silabs-ArjanB, 2020-09-01:
This should do it then: https://github.com/openhwgroup/core-v-docs/pull/179

Silabs-ArjanB, 2020-09-11:
Hi @MikeOpenHWGroup Can this be closed now?

MikeOpenHWGroup, 2020-09-11:
Indeed.  Closing.

closed 2020-09-11 by MikeOpenHWGroup