openhwfoundation/programs #135: Bit manipulation (set and clear) descriptions

DavidM-EMUS, 2020-07-20:
In the documentation, updated descriptions of the functionality of p.bclr, p.bclrr, p.bset, and p.bsetr are unclear and possibly incorrect. The documentation lists the functionality of p.bclr as:
"p.bclr | rD, rs1, Is3, Is2 | rD[min(Is3+Is2,31):Is2] = 0 the rest of the bits of rD are passed through and are not modified" at
https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html#bit-manipulation-operations
This description is confusing and/or misleading for three reasons:
1; If the instruction performs as specified above, what is the purpose of the rs1 argument?
2: The instruction didn't perform in the manner specified above in the past.
3: The instruction doesn't perform as specified in testing. The instruction presently copies rs1 to rD and then performs as specified in the documentation. For example, under the present description,
p.bclr 0x82f98620, 0x12577823, 8, 24
would output 0x00f98620, when testing shows that the result is 0x00577823.

These issues are also the same for p.bclrr, p.bset, p.bsetr.

Recommended fix: Add to description to denote that rD is set to the value of rs1 with the specified bits cleared.
Example p.bclr:
rD = (rs1 & ~(((1<<ls3)-1)<<ls2))
Example for p.bclrr:
rD = (rs1 & ~(((1<<rs2[9:5])-1)<<rs2[4:0]))
Example for p.bset:
rD = (rs1 | (((1<<ls3)-1)<<ls2))
Example for p.bsetr:
rD = (rs1 | (((1<<rs2[9:5])-1)<<rs2[4:0]))

davideschiavone, 2020-11-10:
Hi @DavidM-EMUS ,

I changed the Bit Manipulation in this commit to make it clearer as there were confusion about the meaning, and I have been asked to write them in a more human-readable way

https://github.com/openhwgroup/core-v-docs/commit/1748db6873f123014833cdb34eb523afaccf5af7

So in the clear example I would fix it like:

rD[min(Is3+Is2,31):Is2] = 0. The rests of the bits of rD are equal to the rs1 respective ones.

Otherwise I can undo that commit and modify as you suggested but it's not human readable, and you may miss the purpose. Or we can do a mix, like the bitrev, which contains an example.

Let me know and I will modify the docs accordingly. I would probably prefer the mixed approach

DavidM-EMUS, 2020-11-10:
Hi @davideschiavone,

I also prefer the mixed approach. I am not certain that your proposed fix would resolve all confusion, but I believe that an example in tandem with the fixed description should clear up confusion.

davideschiavone, 2020-11-10:
Actually, thinking about it more, for set and clear the formula is already quite clear. I will just update the spec as you suggested. Thanks

commit 21ee9178 (2020-11-10): :bug: fix #135
commit 4549b71c (2020-11-10): Merge pull request #258 from davideschiavone/fix135

:bug: fix #135
closed 2020-11-10 by davideschiavone