openhwfoundation/programs #139: Incorrect operands for pv.(extract/extractu/insert).(h/b)

DavidM-EMUS, 2020-07-22:
In the SIMD ALU Encoding section of the documentation at
https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/instruction_set_extensions.html#simd-alu-encoding
instructions pv.extract.h, pv.extract.b, pv.extractu.h, pv.extractu.b, pv.insert.h, and pv.insert.b all have their operands listed as "rD, Imm6"
This conflicts with the description and performance of the instructions, in which all require a register, rs1, as well. This can be seen in the description of pv.extract.h:
rD = Sext(rs1[((I+1)\*16)-1 : I\*16])
where I is a stand-in for Imm6.
Changing the operands from "rD, Imm6" to "rD, rs1, Imm6" in the SIMD ALU Encoding section for the instructions listed above should remedy this issue.

GTumbush, 2020-10-27:
This causes test Xpulp test pulp_vectorial_bit_manip  to fail.

davideschiavone, 2020-11-10:
@DavidM-EMUS , @GTumbush , a PR has been done at #257 . If one of you two can review it, I will merge it in the master branch.

DavidM-EMUS, 2020-11-10:
The pull request looks good to me. Thank you for taking care of this @davideschiavone.

commit 7d95d96e (2020-11-10): :bug: fix #139
commit b0e239ea (2020-11-10): Merge pull request #257 from davideschiavone/fix139

:bug: fix #139
closed 2020-11-10 by davideschiavone