openhwfoundation/programs #140: Ambiguous description of uhartid CSR

MikeOpenHWGroup, 2020-07-22:
The description of the User Hardware Thread ID ([uhartid](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html#user-hardware-thread-id-uhartid)) at CSR address 0x014 is ambiguous.  Questions that could be answered in the UM are:

1. How is uhartid different than [mhartid](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html#hardware-thread-id-mhartid)?
2. Why does uhartid only exist when PULP_XPULP = 1?
3. What, exactly, defines the reset value?

In simulation it appears that the power-on-reset value of this CSR is set to `uhart_id_i`.   Is that the intended behavior?

Silabs-ArjanB, 2020-08-31:
Hi @MikeOpenHWGroup This issue should have been solved now with https://github.com/openhwgroup/core-v-docs/pull/178. Could you please close this issue if agreed?

MikeOpenHWGroup, 2020-08-31:
Agreed.  Thanks.  Closing.

closed 2020-08-31 by MikeOpenHWGroup