openhwfoundation/programs #143: CSR TDATA1: incorrect power-on-reset value in the Use Manual

MikeOpenHWGroup, 2020-07-23:
### Type: Factually incorrect information.
The documented PoR for `tdata1` (0x7A1) is 0x2800_1000.   However, a read-after-reset yields 0x2800_1040 (bit 6 is unexpected set).  The description of bit 6 is `m: 1 = Match in M-Mode.`.   Since CV32E40P operates either in M or Debug modes only, its seems to me that the RTL is returning the correct value, and the documentation needs updating.

### Location of Issue
1. [URL](https://github.com/openhwgroup/core-v-docs/blob/master/cores/cv32e40p/user_manual/source/control_status_registers.rst)
2. Git Hash: 008b08f949f5df79b3d26989920bcd37d601219e
3. Section: Control and Status Registers

### Additional context
To reproduce:<br>
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif/cv32/sim/uvmt_cv32
$ make custom CUSTOM_PROG=csr_por USE_ISS=NO WAVES=1
```
Please note that the end of simulation banner incorrectly reports a pass, even through the testcase clearly indicates the failure (scroll up to see it).   We have bugs in testbenches too.  :stuck_out_tongue_winking_eye:

Silabs-ArjanB, 2020-09-21:
The reset value of  0x2800_1040  is okay. The documentation already reflects this. Can this issue be closed?

MikeOpenHWGroup, 2020-09-21:
Yes, it can.  Thanks!

closed 2020-09-21 by MikeOpenHWGroup