openhwfoundation/programs #145: CSR TINFO: unspecified power-on-reset value in the Use Manual

MikeOpenHWGroup, 2020-07-23:
### Type: Confusing or poorly worded information.
The documented PoR for `tinfo` (0x7A4) is not provided.   However, a read-after-reset yields 0x0000_00004.  Based on my interpretation of the User Manual plus the RISC-V privileged and debug specs, I believe this to be the correct value:
* `tdata1[31:28]` reads 4'h2 (trigger data is "type 2")
* `tinfo[15:0]` read 16'h0004 - bit pos 2 is set matching "type 2"

So essentially, `tinfo[15:0]` is a one-hot representation of `tdata1[31:28]`.   This matches what I am seeing in simulation. Given that this is read-only information it would be helpful to be explicit about this in the User Manual.

### Location of Issue and Additional Context
See https://github.com/openhwgroup/core-v-docs/issues/143

jm4rtin, 2020-07-23:
The debug specification is quite open in what features are implemented. In a possible implementation you can actually write `type` to one of multiple types and you can write it different for multiple triggers selected by `tselect`. If multiple trigger types are implemented (which isn't the case in the core right now) then `tinfo[15:0]` is not actually a one-hot representation of `type` . For example if you have match, interrupt and exception trigger types then `tinfo[15:0]` may "reset" to 16'h0034. You can also support different types for each trigger selected with `tselect` and so `tinfo[15:0]` would "reset" to represent the types supported for the first trigger when `tselect` is reset to 0.

> The **info** field contains one bit for each possible _type_ enumerated in _tdata1_. Bit N corresponds to type N. If the bit is set, then that type is supported by the currently selected trigger. If the currently selected trigger doesn’t exist, this field contains 1.

This is actually very close to the wording in the debug specification. It doesn't describe any assumption made during the implementation of this core. Do you prefer it to reiterate which trigger types are supported?

MikeOpenHWGroup, 2020-07-23:
Thanks for the clarification @jm4rtin.  I was a bit off in my understanding.

> This is actually very close to the wording in the debug specification. It doesn't describe any assumption made during the implementation of this core. Do you prefer it to reiterate which trigger types are supported?

Very close indeed - too close, which is the issue.  Neither are clear.   In CV32E40P the PoR of this CSR can never be anything other than 0x4, so the UM could just say that.

Silabs-ArjanB, 2020-09-21:
Added reset value and added comment that type is always type 2. https://github.com/openhwgroup/core-v-docs/pull/197

MikeOpenHWGroup, 2020-09-21:
Thanks!  Closing.

closed 2020-09-21 by MikeOpenHWGroup