openhwfoundation/programs #163: Incomplete description of DCSR

silabs-oysteink, 2020-08-11:
## Bug Title
Description of DCSR should include information about non-implemented bits and their values.

### Type
* Factually incorrect information.

### Location of Issue
https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html#debug-control-and-status-dcsr
Git hash: c6b1e8e2

I guess the table below should indicate that CV32E40P only supports machine mode, thus rendering the ebreaku bit tied to zero, and priv always 3. Stepi should be renamed to stepie.

Bit # | R/W | Description
-- | -- | --
31:28 | R | xdebugver: returns 4 - External debug support exists as it is described in this document.
15 | R/W | ebreakm
12 | R/W | ebreaku
11 | R/W | stepi
8:6 | R/W | cause
2 | R/W | step
1:0 | R | priv: returns the current priviledge mode

Silabs-ArjanB, 2020-09-21:
Fixed reset value of dcsr
Added missing bitfields
Described that ebreaku, ebreaks, prv are implemented as RO/WARL (as in 0.14 spec)

Fix done in https://github.com/openhwgroup/core-v-docs/pull/197

Silabs-ArjanB, 2020-09-21:
Corresponding \rtl fix is merged with https://github.com/openhwgroup/cv32e40p/pull/513

Silabs-ArjanB, 2020-10-22:
Hi @silabs-oysteink , can you please close this one if you agree it has been fixed?

closed 2020-10-22 by silabs-oysteink