openhwfoundation/programs #175: Bits 31:0 of mhpmcounter3 are not 0

darbaria, 2020-08-26:
Bits 31:0 of mhpmcounter3 are not 0, violating the requirement specified in Section Machine Performance Monitoring Counter
All other counters satisfy this requirement. Branch tested 16 July cv32e40p. 

[mhpm_counter3_lsb32_bits_issue.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5123917/mhpm_counter3_lsb32_bits_issue.vcd.gz)

### Component
CSR registers

### Steps to Reproduce
Used formal property checking
[mhpm_counter3_lsb32_bits_issue.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5123921/mhpm_counter3_lsb32_bits_issue.vcd.gz)

Silabs-ArjanB, 2020-08-31:
Hi @darbaria The documentation has been updated. Could you please close this issue if it it clear now (or let us know what is still not clear)?

darbaria, 2020-08-31:
OK

closed 2020-08-31 by darbaria