openhwfoundation/programs #184: value of marchid

MikeOpenHWGroup, 2020-09-09:
The read value of `marchid` in the user manual is 0x0, but the CSR is hardcoded to 0x4 in the RTL.

I assume 0x4 is correct, but why do we have that value?

Silabs-ArjanB, 2020-09-10:
Hi @MikeOpenHWGroup The User Manual was updated 10 days ago with value 4, see https://github.com/openhwgroup/core-v-docs/pull/178

Is the manual not being built automatically?

Furthermore see https://github.com/openhwgroup/cv32e40p/issues/431

MikeOpenHWGroup, 2020-09-10:
Well, this is embarrassing.  I recall #431 and was sure this was an issue that had previously been discussed.  I'm not sure _what_ I was looking at but it was created out of date material.   The UM is fine and I'll drink a cup of coffee to ensure I'm awake.

closed 2020-09-10 by MikeOpenHWGroup