openhwfoundation/programs #186: Read value of CSR mimpid

MikeOpenHWGroup, 2020-09-11:
Currently the read value of `mimpid` is 0x0 and the user manual also says it will be 0x0.   In the RISC-V ISA it says:

```
The mimpid CSR provides a unique encoding of the version of the processor implementation. This
register must be readable in any implementation, but a value of 0 can be returned to indicate that
the field is not implemented.
```

The RTL, user manual and ISA are aligned so on the face of it, we are OK.   Having said that, the IP for the CV32E40P core may have multiple "releases" and if I understand the intent of mimpid correctly, this CSR should be unique for each release.  Of course, the value of mimpid for the first release of CV32E40P could be zero, but that may be confused with "not implemented".

So, the question is, should we leave mimpid as it is (0x0), or should we define it to be something else?  For example, if we set it to 0x1, that could indicate that this is "release 1" of the IP and clearly indicate to the users of the core that we unambiguously meant it to be "1".

Silabs-ArjanB, 2020-09-11:
Hi @MikeOpenHWGroup As you said it is implemented and documented okay. Let's not cause extra work. It will get increased on a 2nd official release.

MikeOpenHWGroup, 2020-09-11:
OK, I'm content with that.

closed 2020-09-11 by MikeOpenHWGroup