openhwfoundation/programs #189: Default parameter configuration for CV32E40P

MikeOpenHWGroup, 2020-09-14:
@Silabs-ArjanB has made the following proposal for parameter defaults in verification, documentation and RTL of the CV32E40P as follows:

```
PULP_XPULP                =  0
PULP_CLUSTER              =  0     
FPU                       =  0
PULP_ZFINX                =  0
NUM_MHPMCOUNTERS          =  0
```
 
This would require an update to the [core integration](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html#core-integration) section of the User Manual (at least).  @Silabs-ArjanB have further suggested that we should add notes/warnings in the User Manual and corresponding assertions in the RTL that PULP_XPULP = 1, PULP_CLUSTER = 1, PULP_ZFINX = 1 have not yet been verified and that the instruction encoding of the PULP instructions will change (in a non-backward compatible manner), as well as that FPU = 1 has not been verified.

For the record, I am strongly in favor of this.

silabs-PaulZ, 2020-09-14:
Hi @Silabs-ArjanB , @strichmo , @MikeOpenHWGroup ,

Can we properly ensure the HPM counters and associated events are verified with the NUM_HPMCOUNTERS=0? If not, then we may need to make this default to 1 (or change the scope of the first release to not include this coverage).

MikeOpenHWGroup, 2020-09-14:
> Can we properly ensure the HPM counters and associated events are verified with the NUM_HPMCOUNTERS=0?

I do not see how.  The question is, does the first release of CV32E40P support HPMCOUNTERS or not?   If it does, then we could make the default NUM_HPMCOUNTERS equal to 1 and add a note to the User Manual saying that NUM_HPMCOUNTERS equal to zero or greater than one has not been verified.

Silabs-ArjanB, 2020-09-15:
>> The question is, does the first release of CV32E40P support HPMCOUNTERS or not? 

Yes, so let's keep NUM_HPMCOUNTERS=1 then if that is what it takes to get it properly verified.

Silabs-ArjanB, 2020-09-21:
Discussed this with @davideschiavone and we agree to change the documentation and RTL as follows to reflect the RTL that will be verified initially:

PULP_XPULP                =  **0**
PULP_CLUSTER              =  0     
FPU                       =  0
PULP_ZFINX                =  0
NUM_MHPMCOUNTERS          =  1

_After initial RTL freeze and once the PULP extension has been verified_, the default will be changed back to in both documentation and RTL as this is the CV32E40**P** core after all:

PULP_XPULP                =  **1**
PULP_CLUSTER              =  0     
FPU                       =  0
PULP_ZFINX                =  0
NUM_MHPMCOUNTERS          =  1

Silabs-ArjanB, 2020-09-21:
Issue has been fixed with https://github.com/openhwgroup/core-v-docs/pull/196

MikeOpenHWGroup, 2020-09-21:
Thanks everyone, I see the updates in the User Manual, and will close this issue now.

closed 2020-09-21 by MikeOpenHWGroup