openhwfoundation/programs #207: Waveforms for LSU protocol are a) double labeled b) have data_rdata_i/data_rvalid_i switched

GTumbush, 2020-09-24:
Two issues in  figures 5,6,7,8 in LSU protocol chapter
1) Figures are double labeled
2) data_rdata_i and data_rvalid_i are switched 

https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/load_store_unit.html#protocol

Interestingly, https://github.com/openhwgroup/core-v-docs/blob/master/cores/cv32e40p/user_manual/source/load_store_unit.rst does not have issue 1)

commit hash: e43c359
[image]

Silabs-ArjanB, 2020-09-25:
Hi @GTumbush Thanks for pointing this out. It should be fixed now with https://github.com/openhwgroup/core-v-docs/pull/208. Can you please close this ticket if agreed?

(I will also fix the table numbering)

Silabs-ArjanB, 2020-09-25:
https://github.com/openhwgroup/core-v-docs/pull/209

GTumbush, 2020-09-25:
Verified as fixed.

closed 2020-09-25 by GTumbush