openhwfoundation/programs #219: Unexpected illegal instruction exception raised on CSR writes

MikeOpenHWGroup, 2020-10-08:
Write accesses to `mvendorid`, `marchid`, `mimpid` and `mhartid` all cause an illegal instruction exception.  This is compliant with the Privileged ISA which states "Attempts to access a CSR without appropriate privilege level or to write a read-only register also raise illegal instruction exceptions".

However, it does not agree with the User Manual with states: "Writes of a non-supported value to a CSR do not result in an illegal instruction exception."

The errant code is easy to see in [cv32e40p_decoder](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/cv32e40p_decoder.sv#L2386).

So either we change the RTL or the User Manual.  My vote is to change the RTL for several reasons:
1. Writing other read-only registers, such as `misa` does not cause an illegal instruction exception.
2. CSR `tdata1` has a single WARL bit (bit pos 2).  Writing to any of the other RO bits does not cause an exception.
3. It seems nonsensical to raise an exception to write a RO register.  The write should simply be ignored.

If we decide to change the RTL to match the User Manual, I can produce a testcase.

Silabs-ArjanB, 2020-10-22:
Hi @MikeOpenHWGroup  , can you please close this one if you agree it has been fixed?

MikeOpenHWGroup, 2020-10-22:
Agree the issue has been resolved.  Closing.   Thanks for the clarification in the spec.

closed 2020-10-22 by MikeOpenHWGroup