openhwfoundation/programs #223: Writes to MCAUSE Exception codes

MikeOpenHWGroup, 2020-10-09:
The bottom 30-bits of the `mcause` CSR is spec'ed to be WLRL.  The CV32E40P RTL is implemented such that the bits [30:5] are read-only, always zero and bits [4:0] are fully read-write.  It is not clear if this is compliant with the ISA because it allows illegal values for the  Exception Code field.  Specifically, the core logic should never violate the following:
```
    mcause[31] && (mcause[1] || mcause[4] || mcause[5] || mcause[9])
    // Supervisor software interrupt , User timer Supervisor timer or Supervisor external interrupt

    !mcause[31] && (mcause[8] || mcause[9])
    // e-call from U- or S-mode
```
In addition, software (csr instructions) should not be able to:
- set either or mcause[1], mcause[4], mcause[5] or mcause[9] and mcause[31] in the same write;
- set either or mcause[1], mcause[4], mcause[5] or mcause[9] if mcause[31] is already set;
- set mcause[8] or mcause[9] when clearing mcause[31] in the same write;
- set mcause[8] or mcause[9] if mcause[31] is already clear.

@eroom1966 has raised an issue against the ISA to seek clarification:
https://github.com/riscv/riscv-isa-manual/issues/597

Imperas, 2020-10-09:
Hi @Silabs-ArjanB 
This is a question of what the RM (and RTL) should do, when the software is not observing the legal behavior.
For example, if we had pseudo random tests, they could write an unsupported/illegal value, in that circumstance what is the defined/expected behavior ?

Thx
Lee

MikeOpenHWGroup, 2020-10-09:
> software should simply not write illegal values

I agree that is what software should not do.   But we are verifying RTL here and eventually software _will_ write an illegal value.  The design needs to be sufficiently robust to handle these situations.

Silabs-ArjanB, 2020-10-12:
I have fixed mcause 'exception code' bitfield width in the manual to 4:0 and added that the other bits read as zero.

For verification purposes: Bit 31 and bits [4:0] are implemented as R/W.

eroom1966, 2020-10-12:
@Silabs-ArjanB 
Hi Arjan
Does this change prevent me from reading an illegal value (following an illegal write)

Thx
Lee

Silabs-ArjanB, 2020-10-12:
Hi @eroom1966 No, the hardware is unchanged. (The behavior you seem to want is that of a WARL bitfield, but this is imply a WLRL bitfield)

Imperas, 2020-10-12:
@Silabs-ArjanB @MikeOpenHWGroup 

I need to implement the desired behavior in the ISS/RM for the circumstance when a non-supported value is written to the Exception-Code (WLRL) field

There are many possibilities including
1. Trap
2. Ignored/discarded
3. Written unconditionally

From the RTL test I ran last week, it appears that (3) is the current implemented behavior
Which I think violates the 'Read Legal' aspect, as I was reading back non-legal values

Thx
Lee

Silabs-ArjanB, 2020-10-12:
Hi @eroom1966 We implement (3) indeed. I do not see anything wrong with that as it is a WLRL bitfield and your scenario is violating the 'WL' part after which the behavior is simply UNDEFINED.

Imperas, 2020-10-12:
@Silabs-ArjanB 
Hi Arjan,
Perfect if that is the design specification, we will ensure that the RM models this as defined.
Thx
Lee

Silabs-ArjanB, 2020-10-12:
Hi @eroom1966 Please note that only bit 31 and bit[4:0] are implemented; the others always read as 0

MikeOpenHWGroup, 2020-10-13:
I do not agree with the direction we are taking here.  In Datacomm ASICs it is considered dangerous to allow software to "write the wrong thing".  In those contexts, the RISC-V field specifications for CSRs (WLRL, etc.) are risky and I thought we had agreed to replace these with the safer RO and RW "modes".  These access modes more clearly define hardware behavior and can be implemented in a way avoids undefined behaviors and maintains compatibility with software that properly follows the field specification rules.

The solution to this issue in pr #227 adds a "WLRL" designation to a mode column.  This is different that every other mode specification in the CV32E40P [CSR](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/control_status_registers.html#csr-descriptions) spec.  I maintain that we should not be adding it now (or ever).

Another way to implement and specify the access mode of `mcause` that is compatible with the WLRL field specification and the current behavior of the CV32E40P RTL is as follows:
```
+-------------+-----------+----------------------------------------------------------------------------------+
|   Bit #     |   Mode    |   Description                                                                    |
+=============+===========+==================================================================================+
| 31          |   RW      | **Interrupt:** This bit is set when the exception was triggered by an interrupt. |
+-------------+-----------+----------------------------------------------------------------------------------+
| 30:5        |   RO (0)  | Always 0                                                                         |
+-------------+-----------+----------------------------------------------------------------------------------+
| 4:0         |   RW      | **Exception Code**   (See note below)                                            |
+-------------+-----------+----------------------------------------------------------------------------------+
**NOTE**: software accesses to mcause[4:0] must be sensitive to the WLRL field specification of this CSR.  For example,
when mcause[31] is set writing 0x1 to mcause[1] (Supervisor software interrupt) will result in UNDEFINED behavior.
```

We could also define W1C and W0C access modes and fully specify the behavior of mcause[4:0] for the situations where mcause[31] is set and cleared.  This would take considerably more spec-ing and a few RTL changes as well and so I will not push for this now.

Silabs-ArjanB, 2020-10-13:
Hi @MikeOpenHWGroup I am okay with your above spec change as well if you make a pull request for that. The original RO mode was just wrong for bits 31 and 4:0. We are simply sticking to the offical RISC-V spec and we have no intention to change the RTL to prevent users from writing wrong values into WLRL bitfields.

MikeOpenHWGroup, 2020-10-13:
>  I am okay with your above spec change as well if you make a pull request for that.

OK, will do.

>  we have no intention to change the RTL to prevent users from writing wrong values into WLRL bitfields

Well, you are the chair of the CTG, so its your call.  IMHO, the concept of RISC-V CSR Field Specifications is flawed.  A robust implementation of proper access modes that is compatible with Field Specifications would be a benefit to the ecosystem and a differentiator for CORE-V.

Imperas, 2020-10-13:
I think cv32e40p should ALWAYS have a defined behavior that is in keeping
the riscv.org specs - and if riscv.specs say something is
implementation defined, (ie like 'undefined') then openhw should define
what its core does.

there is no place in the verification world for undefined...

define it.

:-)

-- 
====================================================================
The information contained in this electronic mail message and any
attachments hereto
is privileged and confidential information intended only for the use of the
individual or
entity named above or their designee.  If the reader of this message is not
the intended
recipient, you are hereby notified that any dissemination, distribution or
copying of this
communication is strictly prohibited.  If you have received this
communication in error
please immediately notify us by return  message or by telephone and delete
the
original message from your mail system.  Thank you.
====================================================================

eroom1966, 2020-10-14:
Hi @Silabs-ArjanB  

> Hi @MikeOpenHWGroup I am okay with your above spec change as well if you make a pull request for that. The original RO mode was just wrong for bits 31 and 4:0. We are simply sticking to the offical [...]

In which case any value written to MCAUSE[31], MCAUSE[4:0], will be read back **unmodified** (assuming no state change from any generated exception)

thx
Lee

Silabs-ArjanB, 2020-10-14:
Hi @eroom1966 Correct, that is what will happen (which is fully within the RISC-V privileged spec)

MikeOpenHWGroup, 2020-10-14:
OK, I think we are all aligned, if maybe not all fully content.  :wink:

Insofar as this issue is concerned, the RM and RTL are functionally equivalent and I have a passing simulation to demonstrate that.

Unless anyone objects, I'll close this issue as soon as #228 is merged in.

eroom1966, 2020-10-14:
no objections

Silabs-ArjanB, 2020-10-22:
Hi @MikeOpenHWGroup  , can you please close this one if you agree it has been addressed?

closed 2020-10-22 by MikeOpenHWGroup