openhwfoundation/programs #232: Add to cv32e40p ISA verification plan to cover every instr->every instr

GTumbush, 2020-10-26:
## Bug Title
Add to cv32e40p ISA verification plan to cover every instr->every instr

### Type
Recommend to add to the cv32e40p ISA verification plan a requirement to cover that every instruction has been followed by every other instruction. 

### Location of Issue
Looked through the verification plans at https://github.com/openhwgroup/core-v-docs/tree/master/verif/CV32E40P/SimulationVerificationPlan/base_instruction_set and did not see this cross in the plans.

MikeOpenHWGroup, 2020-10-27:
Good catch @GTumbush, this is indeed a holes in our DV plans.  Having said that, I worry about the cost/benefit, particularily in light of the ISA formal verification work.  So I suggest we proceed in a step-wise fashion.  To that end, please:

1. update and issue a pull-request against the [CV32E40P RV32I DV plan](https://github.com/openhwgroup/core-v-docs/blob/master/verif/CV32E40P/SimulationVerificationPlan/base_instruction_set/CV32E40P_RV32I_Base_Instructions.xlsx) to add the requirement to coverage all legal RV32I instructions followed by all legal RV32I instructions.
2. update and issue a pull-request against the [CV32 functional coverage model](https://github.com/openhwgroup/core-v-verif/blob/master/cv32/env/uvme_cv32/cov/uvme_cv32_cov_model.sv) to add this functional coverage.

Once this is in place we can quickly review the impact to coverage and make a determination about whether to expand this across all supported instructions.

GTumbush, 2020-10-30:
Access to GitHub.com has been restored. Will get on this ASAP.

GTumbush, 2020-10-30:
Done:
1. Pull request https://github.com/openhwgroup/core-v-docs/pull/241
2. Pull request https://github.com/openhwgroup/core-v-verif/pull/320

Please let me know when I can close this issue.

MikeOpenHWGroup, 2020-11-03:
Hi @GTumbush, we can close this issue now since both gating PRs have been merged in.   If you agree, go ahead and close it.

GTumbush, 2020-11-03:
Plan and code updates merged. Closing.

closed 2020-11-03 by GTumbush