openhwfoundation/programs #245: Illegal instruction conditions for CSR access instructions

MikeOpenHWGroup, 2020-11-04:
Looking at [cv32e40p issue #565](https://github.com/openhwgroup/cv32e40p/issues/565) I see a potential source of confusion for users because the design makes a decision that is not explicitly specified by the ISA.

It seems the ISA does not specify what else to do if an otherwise properly formed CSR access instruction was issued where instr[13:12] == 2'b00, other than gating any instruction side-effects before they happen.  The designer chose to flag an illegal instruction, which is probably the most expedient approach since there is already logic in the pipeline to handle illegal instructions.  I agree with the outcome of [cv32e40p issue #565](https://github.com/openhwgroup/cv32e40p/issues/565) that the CV32E40P is compliant with the ISA in this regard.

Since the ISA is not explicit about what happens, _we_ should be.  It would be appropriate to update the User Manual to highlight this, probably in the CSR chapter (but I leave it to your better judgement as to location in the UM).

Silabs-ArjanB, 2020-11-05:
We aim to treat the non-defined instructions that 'look like' CSR instructions (but which are simply not CSR instructions) the same as any other non-defined instructions. We cause an illegal instruction exception. This is allowed as explicitly mentioned in the Unprivileged spec: 

[image]

The only potential issue I see here that I am not aware that on the verification side all undefined encoding are checked, so I would rather not start listing specific illegal encodings in the User Manual as it would be a huge job. We can also not give a blanket statement that all undefined encoding will trap as we are likely not verifying that. Either way, we are fully within the spec here.

MikeOpenHWGroup, 2020-11-05:
> We aim to treat the non-defined instructions that 'look like' CSR instructions (but which are simply not CSR instructions) the same as any other non-defined instructions. We cause an illegal instruction exception. 

I completely agree with this strategy and it has always been my understanding that that is the case.  I think we are _almost_, but not completely, compliant with the unprivileged spec.  The spec says "Instruction encodings that are not defined here ... may cause a fatal trap."   We are throwing illegal instruction exceptions, which I believe are invisible traps, not fatal traps.

You are right that we should not specifically call out the unspecified encoding of CSR instructions.  Rather, a more general statement such as the following might be better:
```
"Instruction encodings that are not defined here and not defined in the ISA will cause an illegal instruction exception.
```

As to your other question, David McConnell of em-micro and I are working on a testcase that is intended to hit all undefined encodings.  This is somewhat of an unbounded task, so our completion metric is to exercise all cases in the decoder that assert illegal_instruction (I think there are about 50 of these if the PULP decoder is excluded).  I agree that these should not all be listed in the UM.

Silabs-ArjanB, 2020-11-05:
Hi @MikeOpenHWGroup Can you explain what you consider the difference to be between an invisible trap and a fatal trap?

The directed test that you are describing sounds a little like a self fulfilling prophecy. Ideally such a test should take the RISC-V spec as its starting point as opposed to the RTL, unless your only goal is to increase RTL coverage of course.

Silabs-ArjanB, 2020-11-05:
Hi @MikeOpenHWGroup 

The statement ""Instruction encodings that are not defined here and not defined in the ISA will cause an illegal instruction exception."" would not be proven by the testcase that you describe as that test case seems to use the RTL as its 'spec' and is therefore more of a coverage driven test. Before we can claim the quoted statement we should verify that all undefined instructions are treated as illegal.

With respect to being 'no completely compliant' I don't see a difference between invisible traps and fatal traps from the RTL point of view; isn't that just a software notion?

MikeOpenHWGroup, 2020-11-05:
> Ideally such a test should take the RISC-V spec as its starting point as opposed to the RTL

Well _of course_ we are doing that.  We are simply using the RTL code coverage as the completion metric.

> Before we can claim the quoted statement we should verify that all undefined instructions are treated as illegal.

No, and you know better.  The Specification details intent, not "what is verified".  The spec comes first.

> With respect to being 'no completely compliant' I don't see a difference between invisible traps and fatal traps from the RTL point of view; isn't that just a software notion?

Good question.  I always read specs as if they are hardware specs, so the statement "[fatal] trap represents a fatal failure and causes the execution environment to terminate execution" says to me that the RTL stops the pipeline.  AFAIK, we do not implement this which is why I asked for the clarification in the UM.   Having said that, you may be right, from the RTL perspective, there may be no distinction between and "invisible trap" and a "fatal trap".

I still see value adding the clarification in the UM.  It clarifies our intent on instruction decode and avoids the "invislble trap vs. fatal trap" question.

Silabs-ArjanB, 2020-11-05:
>> No, and you know better. The Specification details intent, not "what is verified". The spec comes first.

Agreed with that

davideschiavone, 2020-11-06:
I agree than adding a single sentence to clarify this behaviour is safer, and creates less confusion for future users. I think we are compliant anyway, but adding a single sentence does not hurt. Of course we cannot write each single instruction that causes an illegal traps, as it is a huge , huge job :) Adding the sentence right now

MikeOpenHWGroup, 2020-11-06:
Thanks guys, fixed with #249.

closed 2020-11-06 by MikeOpenHWGroup