openhwfoundation/programs #259: Implementations does not raise illegal instruction exceptions on machine-mode access to debug registers

darbaria, 2020-11-11:
## Bug Title

According to the ISA: Machine-mode standard read-write CSRs 0x7A0–0x7BF are reserved for use by the debug system. Of these CSRs,0x7A0–0x7AF are accessible to machine mode, whereas0x7B0–0x7BF are only visible to debug mode. [Implementations should raise illegal instruction exceptions on machine-mode access to the latter set of registers.](url
[MACHINE_MODE_ACCESS_TO_DEBUG_ADDR_DOES_NOT_RAISE_IILEGAL.vcd.gz](https://github.com/openhwgroup/cv32e40p/files/5401410/MACHINE_MODE_ACCESS_TO_DEBUG_ADDR_DOES_NOT_RAISE_IILEGAL.vcd.gz)
)

Found by:
App: formalISA
Assertion failure
In the attached VCD you can see csr_addr is inside {['h7B0:'h7BF] on a valid CSR decode but the illegal insn is not raised.

MikeOpenHWGroup, 2020-11-11:
Transfered from cv32e40p as @darbaria recommends that this be discussed as a doc'n issue.

darbaria, 2020-12-07:
We discussed it. It seems that Davide's concern was that we were not checking for exception updates, but we were. So we close it.

closed 2020-12-07 by darbaria