openhwfoundation/programs #261: Add description about Core Versions and RTL Freeze

davideschiavone, 2020-11-15:
CV32E40P has its own architectural id **_marchid_** that defines its architecture, ISA, and features.
This value never changes in the CV32E40P repository, and the RISC-V foundation gives it.
For every change in the architecture or ISA, it is highly recommended to create a new repository and ask for a new **_marchid_**
rather than changing the pre-agreed features of the core.

Its implementation id **_mimpid_** specifies which parameters are verified on a given date, which we refer to as "RTL Freeze, " labeled by a **_git tag_**.

Not all the parameters are verified at a given time. In addition, bugs can always be found after "RTL Freeze," even on pre-verified parameter values. On top of that, power, performance, area (PPA) optimizations can be implemented in the core.

How can we control the CORE version?

1. The RTL Freeze is controlled by the (mimpid, **git tag**, verified parameters)  set (rtl-freeze-set).
Only Logically-Equivalent changes are allowed within this set. A new git tag is defined.

2. A bug is found: by definition, its solution will be not logically equivalent. Thus a new rtl-freeze-set is defined by incrementing the mimpid in the RTL (**mimpid**, **git tag**, verified parameters). The documentation should report such bug specifying which rtl-freeze-set fails it and which not. A new git tag is defined. 

3. New parameters are verified, and changes in the RTL are required. Thus, a new rtl-freeze-set is defined by incrementing the mimpid in the RTL and the documentation's verified parameters (**mimpid**, **git tag**, **verified parameters**). The documentation should report the new parameters verified with respect to the mimpid. Every change in the new parameters must be logically-equivalent to the pre-verified parameters. For example, if a change in the RTL is required for a PULP instruction (i.e., XPULP parameter is equal to 1), such change must be logically equivalent when the XPULP parameter is set to 0. 

4. Non-Logically-Equivalent PPA optimizations on an rtl-freeze-set are forbidden (e.g., a faster divider is not allowed).

[image]

davideschiavone, 2020-11-15:
@MikeOpenHWGroup , @Silabs-ArjanB , @jm4rtin , can you please add comments about this TASK Description? Shall we add more information, anything missing? I will update the documentation according to the discussion we have here

MikeOpenHWGroup, 2020-11-17:
This is really a great addition to the User Manual @davideschiavone.  My thoughts:

> Not all the parameters are verified at a given time. In addition, bugs can always be found after "RTL Freeze," even on pre-verified parameter values. On top of that, power, performance, area (PPA) optimizations can be implemented in the core.

I think this is already covered [here](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html#parameters) in the User Manual's "Core Integration" chapter.

> How can we control the CORE version?

This could be the heading of a sub-section of the User Manual.  I would recommend changing it to "How do we control..."

> The RTL Freeze is controlled by the (mimpid, git tag, verified parameters) set (rtl-freeze-set). Only Logically-Equivalent changes are allowed within this set. A new git tag is defined.

This is a little confusing.  A git tag is, by definition static, so _no_ changes, logically equivalent or otherwise are allowed on a tag.  It should be possible to have multiple logically equivalent RTL freeze versions of a particular core, but each would have its own tag.

The rest of it I like and recommend it be included in the User Manual.   We should also consider a "release version table" in the User Manual for CV32E40P releases, similar to a document release table.  Maybe something like:

| Git Tag | Date | Reason for Release | Comment |
|-----------|--------|--------|---------|
| cv32e40p_20201123 | 2020-11-23 | Initial Release | See Core Integration for more information |
| cv32e40p_20201225 | 2020-12-25 | LEC change to remove dead code | Just in time for the holidays |

jeremybennett, 2020-11-18:
These are all hardware tags, so not reflected in the ELF header. At a general level the GNU and LLVM software tool chains will work. However there may be impacts on the ELF RISC-V properties section.  There is also a question of whether we need a CORE-V indicator in the ELF header, and if so how we add that.

MikeOpenHWGroup, 2020-11-18:
> These are all hardware tags, so not reflected in the ELF header.

Not my area of expertise, but I would assume (hope?) that we could use the marchid and mipmid in the ELF header.   The value of these CSRs are directly and uniquely related to the tag.  Now that I think of it, perhaps the format for the tag could be:
```
cv32e40p_marchid_mimpid_yyyymmdd
```

jeremybennett, 2020-11-19:
@MikeOpenHWGroup Sorry I wasn't clear. The fields in the ELF header are standardized and there is no place for this information. Wikipedia describes the 21 fields, which occupy 64 bytes. There aren't even enough bits to record extensions being used. Which is why we have the `.riscv.attributes` section. I'm trying to find where that is actually documented (it's not in the PSABI).

davideschiavone, 2020-11-19:
> This is really a great addition to the User Manual @davideschiavone. My thoughts:
>
>
> I think this is already covered [...]

I like the last table, but we should distinguish between functional changes and not.

I propose a table with what a specific mimpid implements/verifies.
And another table with the content of your table, plus the mimpid as well.

Software changes between impid 0 and 1, as a bug may be found.
Software should not if impid is the same, regarldess of optimizations in the core

Silabs-ArjanB, 2020-12-01:
With respect to the tag I propose that we follow the semantic versioning guidelines as recommended by github (https://semver.org/). The upcoming RTL freeze would then become cv32e40p_1.0.0

MikeOpenHWGroup, 2020-12-01:
I'm not strongly opinionated, but do prefer the cv32e40p_1.0.0 style semantics as well.   I would suggest that whatever tagging semantics we adopt also be used in the core-v-docs directory name for the sign-off checklists.  Currently it is [core-v-docs/program/milestones/CV32E40P/RTL_Freeze_marchid4_mimpid0](core-v-docs/program/milestones/CV32E40P/RTL_Freeze_marchid4_mimpid0).  So if the tag is `cv32e40p_1.0.0` perhaps we could rename the directory to `core-v-docs/program/milestones/CV32E40P/RTL_Freeze_1.0.0`, in which case the README could indicate the mimpid of the core.

Silabs-ArjanB, 2020-12-01:
I named the tag: cv32e40p_v1.0.0

PR #303 (2020-12-09): added Core versions and RTL Freeze rules section
This should fix #261
closed 2020-12-09 by davideschiavone