openhwfoundation/programs #265: Add Illegal conditions for instructions within a HWloop

davideschiavone, 2020-11-18:
If the core is executing a hardware loop, instructions like:

Uncoditional jumps, conditonal branches, system calls like mret, dret, fences, and wfi are not allowd.

These constraints are alreayd implemented in the RTL. However, they are not in the _HWloop constraints_ Section of the document

https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/pulp_hw_loop.html#hwloop-specs

MikeOpenHWGroup, 2020-11-18:
What should happen if the core attempts to execute one of these jumps/branches/syscalls in a HWloop?   Perhaps an illegal instruction exception?

davideschiavone, 2020-11-18:
Hi @MikeOpenHWGroup , yes!

jeremybennett, 2020-11-19:
As noted in the PR #261, this affects the compiler teams!

davideschiavone, 2020-12-04:
Closed by PR #266

closed 2020-12-04 by davideschiavone