openhwfoundation/programs #267: The core interface is not updated with the last debug-realted ports

davideschiavone, 2020-11-19:
The core does not have specifically the newest added ports:

debug_havereset_o
debug_running_o
debug_halted_o

They should also be documented in the debug section of the documentation

MikeOpenHWGroup, 2020-11-19:
Hi @davideschiavone, the core _does_ have these outputs.

Silabs-ArjanB, 2020-11-19:
@silabs-PaulZ This has all been documented; could you just review https://github.com/openhwgroup/cv32e40p/pull/564

davideschiavone, 2020-11-19:
> 
> 
> Hi @davideschiavone, the core _does_ have these outputs.
> 
> I agree that they should be mentioned in both the Core Integration and Debug sections of the User Manual and that this should gate Functional RTL Freeze.

The core "does not" in the documentation I mean :)

davideschiavone, 2020-11-19:
Thanks @Silabs-ArjanB - we should merge this PR asap

davideschiavone, 2020-11-19:
Please @silabs-PaulZ , let us know about the revision to close this issue

silabs-PaulZ, 2020-11-19:
I added review comments in #242 and should be ready to close.

Silabs-ArjanB, 2020-11-20:
https://github.com/openhwgroup/core-v-docs/pull/242 has been reviewed and merged now

davideschiavone, 2020-11-20:
Thanks @Silabs-ArjanB and @silabs-PaulZ

MikeOpenHWGroup, 2020-11-20:
It might also be a good idea to update the [instantiation template](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html#instantiation-template) of the Core Integration chapter.

Silabs-ArjanB, 2020-11-20:
? That is a good idea and that is why we did it

MikeOpenHWGroup, 2020-11-20:
I must have had a stale view of the User Manual on RTD.   Looks fine now.

closed 2020-11-20 by MikeOpenHWGroup