openhwfoundation/programs #272: Purpose of use of scan_cg_en_i not clear

MikeOpenHWGroup, 2020-11-21:
The User Manual makes two references to primary input `scan_cg_en_i`:
- The [Interfaces](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html#interfaces) table in core integration states: "Scan clock gate enable".
- In the Getting Starting section the UM references scan_cg_en_i in the [clock gating cell](https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/getting_started.html#clock-gating-cell) sub-section, but goes not explicitly state what the use _should_ to with the input.

Neither of these references makes it clear to the user of the core what should be done with this input.  I recommend that Core Integration chapter have an explicit recommendation for driving this input.

It is my understanding that setting `scan_cg_en_i` forces the clock to the register file to be always enabled, but if `scan_cg_en_i` is de-asserted then the WB stage will enable reg-file clock as needed.  Is that correct?

Note that the verification environment has `scan_cg_en_i` hardwired to 1'b0.

davideschiavone, 2020-11-24:
can this be closed now? @MikeOpenHWGroup ?

MikeOpenHWGroup, 2020-11-24:
Yes it can.  Thanks!

closed 2020-11-24 by MikeOpenHWGroup