openhwfoundation/programs #441: misa CSR is documented as read/write - should be read-only

jeremybennett, 2021-08-19:
https://cv32e40p.readthedocs.io/en/latest/control_status_registers/#cs-registers lists `misa` as `MRW`, it should be `MRO`

### Type
Indicate whether the type of problem you found:
* Factually incorrect information.

### Location of Issue
Please provide:
1. URL to the document: https://cv32e40p.readthedocs.io/en/latest/control_status_registers/#cs-registers

### Additional context
None

Silabs-ArjanB, 2021-08-19:
The table you refer to matches the RISC-V privileged spec and is correct to list misa as MRW.

The misa bitfields are indeed read-only as documented in https://cv32e40p.readthedocs.io/en/latest/control_status_registers/#machine-isa-misa

jeremybennett, 2021-08-19:
> The table you refer to matches the RISC-V privileged spec and is correct to list misa as MRW.
> 
> The misa bitfields are indeed read-only as documented in https://cv32e40p.readthedocs.io/en/latest/control_status_registers/#machine-isa-misa

Hi @Silabs-ArjanB ,

Thanks. This is not stated in the preamble to the table. If you search for CSRs in the document it takes you to this section. There is nothing to say that the table is duplicating the official standard and that the details for individual registers may be different.

I imagine anyone looking up like me would see `misa` is marked as MRW and have no reason to realize that there is another paragraph which tells you that all the individual fields are RO for CV32E40P, so by implication the register is actually MRO.

So I don't believe this issue is resolved. The preamble paragraph should make it clear that this is the situation.

It would be helpful to have hot links in the table to the individual specs below, but that is a nice to have, not needed to resolve this issue.

Thanks, Jeremy

Silabs-ArjanB, 2021-08-19:
Hi @jeremybennett , the meaning of RW and RO is explained above the table. I agree that we could/should add a statement about that access to bitfields is further detailed for the individual CSRs.

jeremybennett, 2021-08-19:
OK - thanks