openhwfoundation/programs #459: SDK: "Installer" (or better "SDK Package Manager") should be the key component

ruspl-afed, 2021-09-24:
Currently SDK Project concept proposal https://github.com/openhwgroup/core-v-docs/blob/master/program/Project%20Descriptions%20and%20Plans/SDK/sdk-project-concept.md contains the "installer" component at the end of the list, while according to "key objectives" it should be the main one and the first one.

Moreover, "Installer" assumes "one time" interaction, that cuts the experience dramatically. But in reality, "S" that means "Software" implies continuous updates. So, it should be a "Package Manager" to form the corresponding viewpoint for requirements gathering.

HughOKeeffe1, 2021-09-24:
Jeremy is working on applying some updates including some that I have provided which hopefully should make things clearer.

jeremybennett, 2021-09-26:
Hugh has provided me with the following wording for Component 8, which I think captures the spirit more cleanly

> The SDK is the integration of all the previous components combined with an installer. The goal for the Installer is that it is provided as a single, binary executable file which when executed installs everything ready-to-run. Everything will be included in the single installer binary (including the IDE, Debugger, Toolchains, Documentation and Examples etc.). The user simply downloads the file from the OpenHW SDK webpage and executes it to install everything. Specific flavours will be needed for Windows, Linux (various) and MacOS host OSs.
>
> SDK updates will be distributed via new versions of the installer and support for a package manager or component only updating is not proposed as this stage.

I agree that the installer is an important element. But I'm not suggesting priorities for any one component, since it leads to the mistaken view that you can cut costs by leaving things out.  We've already done that analysis (no HAL, no CORE-V specific tool chain, no Clang/LLVM). What is left is the bare minimum for a meaningful SDK, and all elements must be delivered.

ruspl-afed, 2021-09-26:
OK, let's forget about "priorty" aspect, I do not insist on it. Let's focus on a scenario. 
As far as I understood, to install an SDK, one needs to download single executable (I assume it will be a number of gigabytes at minimum, but may be several times more) and install everything at once. Very good. Next we release an update for a component for whatever reason. And then what? Should one download another single binary of many gigabytes and then reinstall all again?

jeremybennett, 2021-09-26:
It's a good point, and I'll defer to @HughOKeeffe1 for his thoughts on how an update would work.