openhwfoundation/programs #460: SDK: "Installer" component: improve statement regarding overlap with IDE

ruspl-afed, 2021-09-24:
Current formulation of SDK proposal https://github.com/openhwgroup/core-v-docs/blob/master/program/Project%20Descriptions%20and%20Plans/SDK/sdk-project-concept.md
"There wil be be some overlap with the IDE component, since Eclipse CDT has its own installer." sounds a bit fuzzy.

1) Eclipse CDT "per se" has no installer
2) There is a package "Eclipse IDE for C/C++ Developers" that has Eclipse Oomph installer
3) There is a package "Eclipse IDE for Embedded C/C++ Developers" that has Eclipse Oomph installer
4) There is a CORE-V IDE distribution that contains features from 2) and 3) and does not have a configuration for Eclipse Oomph installer yet
5) We are falling back to issue #459 since we should talk about "package manager"

Although Eclipse ecosystem has enough building blocks to fulfill install/update experience for both UI and headless components I must mention the technologies provided by PlatformIO that already supports "out-of-the-box" experience for embedded development.

HughOKeeffe1, 2021-09-24:
Jeremy is working on applying some updates including some that I have provided which hopefully should make things clearer.

jeremybennett, 2021-09-26:
Thanks for the clarification. I think the wording in [#459](https://github.com/openhwgroup/core-v-docs/issues/459) captures the position regarding the installer. This is only Concept Phase proposal, specifying what we want, the detail of how we will do it is for the Launch Phase.