openhwfoundation/programs #506: Should CVE20 be CV32E20?

MikeOpenHWGroup, 2022-02-11:
There is a project directory called [CVE20](https://github.com/openhwgroup/core-v-docs/blob/master/program/Project%20Descriptions%20and%20Plans/CVE20) in this repository.  Given that this particular core has virtually no chance of being anything other than a 32-bit core, I suggest that we rename is `CV32E20`.

I am creating an issue rather than a pull-request because there may be a good reason for selecting `CVE20` that I am not aware of.

DBees, 2022-02-11:
I agree. The project name should be CV32E20. I will do a pull request to change that. Note that I am shortly to publish a spreadsheet to the community with suggested project names. In general for cores projects, the project names are either the family name, where the project is creating or clearly initiating a family, or a part name where the focus is on the specific part (at least for the current time).

rickoco, 2022-02-11:
@DBees @MikeOpenHWGroup as with the CVA5, CVA6 recent discussions, IMHO, we should stay with CVE2(0). The Family name consistency is something we should striving for with perhaps multiple cores as we have under CVA6 (and in retrospect should have had under CVE4).

MikeOpenHWGroup, 2022-02-11:
Maybe.

The CVA6 and CVA5 both started their life at OpenHW as single repos that could support multiple products.  CV32E40P is not like that.  What I mean is that we could not have a repo called CVE4 from which we could create CV32E40P, CV32E40X and CV32E40S.  (Yes, it would be _possible_, but also so impractical that nobody would do it).

In that context, the CVE2 just doesn't fit. It is always going to be a 32-bit core and that is an important feature that the repo name should make clear.

We _could_ go with CV32E2.

rickoco, 2022-04-25:
We have decided to move fwd with CVE2

closed 2022-04-25 by rickoco